ARMmbed/mbed-os-example-wifi #9: K64F + ARM Compiler 5 + ESP8266 fails the get request to "www.arm.com"

theotherjimmy, 2016-12-08:
using mbed-os 5.3.0-rc3

LOG:
```
WiFi example

Scan:
0 networks available.

Connecting...
Success

MAC: 18:fe:34:dc:dd:7d
IP: 10.0.0.137
Netmask: 
WiFi example

Scan:
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:37:81 RSSI: -56 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4a:3f:20 RSSI: -76 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:3f:21 RSSI: -76 Ch: 1
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -45 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4c:a2:80 RSSI: -88 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4c:a2:81 RSSI: -88 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4a:37:80 RSSI: -56 Ch: 1
Network: AMEREDEV secured: WPA2 BSSID: 38:FF:36:2c:6:18 RSSI: -93 Ch: 4
Network: AccessNG secured: None BSSID: 94:B4:F:4a:40:60 RSSI: -86 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:40:61 RSSI: -84 Ch: 6
Network: AccessNG secured: None BSSID: 94:B4:F:4a:3f:40 RSSI: -76 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:3f:41 RSSI: -76 Ch: 6
Network: AccessNG secured: None BSSID: 94:B4:F:4a:2e:20 RSSI: -78 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:2e:21 RSSI: -78 Ch: 6
Network: IoTBU-Sniffer secured: WPA/WPA2 BSSID: 20:C:C8:4b:2a:ed RSSI: -51 Ch: 1
15 networks available.

Connecting...
Success

MAC: 18:fe:34:dc:dd:7d
IP: 10.0.0.137
Netmask: 
Gateway: 
RSSI: 0

Sending HTTP request to www.arm.com...
sent -3012 [GET / HTTP/1.1]
```

bulislaw, 2016-12-09:
The gateway and netmask are not set, I did the same testing on my side and I was able to get all the network details using ARMC5. But then the ESP seem to be a little bit moody as to which networks ti works with eg it works fine every time using my iphone tethered network, but it fails every now and again using office test network (and when it starts to fail it fails couple of times in a row).

0xc0170, 2016-12-12:
@theotherjimmy Can you retest on a different network?

theotherjimmy, 2016-12-12:
I tried another network:
```WiFi example

Scan:
Network: AccessNG secured: None BSSID: 94:B4:F:4c:a2:20 RSSI: -89 Ch: 1
Network: PhilipsACCF235B5014 secured: WPA2 BSSID: AC:CF:23:5b:50:14 RSSI: -91 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4a:3f:40 RSSI: -58 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:3f:41 RSSI: -58 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4a:3d:0 RSSI: -88 Ch: 1
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -45 Ch: 1
Network: IoTBU-Sniffer secured: WPA/WPA2 BSSID: 20:C:C8:4b:2a:ed RSSI: -45 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4a:37:80 RSSI: -44 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:37:81 RSSI: -44 Ch: 6
Network: AccessNG secured: None BSSID: 94:B4:F:4a:2e:20 RSSI: -55 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:2e:21 RSSI: -57 Ch: 6
Network: AccessNG secured: None BSSID: 94:B4:F:4a:3f:20 RSSI: -75 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:3f:21 RSSI: -73 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 4:BD:88:5a:9a:1 RSSI: -58 Ch: 1
Network: AccessNG secured: None BSSID: 94:B4:F:4c:a2:0 RSSI: -86 Ch: 1
15 networks available.

Connecting...
Success

MAC: 18:fe:34:dc:dd:7d
IP: 192.168.1.95
Netmask: 
Gateway: 
RSSI: 0

Sending HTTP request to www.arm.com...
sent -3012 [GET / HTTP/1.1]

```

mray19027, 2017-06-08:
This problem still exists with HEXIWEAR and ESP8266

Compiler: Online compiler
Imported branch: https://github.com/ARMmbed/mbed-os-example-wifi/tree/mbed-os-5.5.0-rc1-oob

```
WiFi example

Scan:
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fe:81 RSSI: -65 Ch: 1
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fe:80 RSSI: -65 Ch: 1
Network: AccessNG secured: Unknown BSSID: 70:3A:E:55:7d:c0 RSSI: -80 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:55:7d:c1 RSSI: -80 Ch: 1
Network: ESP_1CCF60 secured: None BSSID: 5E:CF:7F:1c:cf:60 RSSI: -48 Ch: 6
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -47 Ch: 6
Network: ESP_DC83BA secured: None BSSID: 1A:FE:34:dc:83:ba RSSI: -56 Ch: 6
Network: GORDON-AP secured: None BSSID: B8:27:EB:d1:f9:c1 RSSI: -53 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:d3:e0 RSSI: -90 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:d3:e1 RSSI: -90 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:c0:61 RSSI: -67 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fc:60 RSSI: -85 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fc:61 RSSI: -83 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:c0:60 RSSI: -69 Ch: 6
Network: IoTBU-Sniffer secured: WPA/WPA2 BSSID: 20:C:C8:4b:2a:ed RSSI: -49 Ch: 1
15 networks available.

Connecting...
Success

MAC: 18:fe:34:f9:e8:20
IP: 192.168.1.39
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: -49

Sending HTTP request to www.arm.com...
sent -3012 [GET / HTTP/1.1]
```

ashok-rao, 2017-06-12:
@0xc0170 .. @geky ..can you please have a look in @bulislaw's absence? Thanks.

deepikabhavnani, 2017-06-13:
I got similar issue with Hexiwear and it fails in connect (response to connect is not captured in test). While performing dns query, response from DNS server (dns_scan_response) is "Format Error".

deepikabhavnani, 2017-06-13:
Issue can be tracked in mbed-os issue [mbed-os/issues/3926](https://github.com/ARMmbed/mbed-os/issues/3926)

0xc0170, 2017-06-14:
> Commit: 984e87d8e Resolves the issue.

If that is case, that patch was included in rc1. Not resolving it at least for https://github.com/ARMmbed/mbed-os-example-wifi/issues/9#issuecomment-307159237 ?

deepikabhavnani, 2017-06-14:
@0xc0170 example is on mbed-os-5.4. We can close this issue when it is on release which has all fixes.

c1728p9, 2017-06-27:
Does 5.5.1 have these fixes? If so you might be able to close it out.

deepikabhavnani, 2017-06-27:
@theotherjimmy I am not getting option to close this. Can you close this?

theotherjimmy, 2017-06-27:
Sure no problem

closed 2017-06-27 by theotherjimmy