ARMmbed/mbed-os-example-wifi #10: Example doesn't print out succesfully to serial

jankii01, 2016-12-12:
The serial prints do not work, in my humble opinion. Just tried it with UBLOX and it only prints out the serial number, nothing else.

The attached example works.

2ndly, I think the default WiFi definitions for more boards should be added - mbed Client has more of those defined.

jankii01, 2016-12-12:
Ah, can't attached .CPP-files. Would need to fork etc. :-/

jankii01, 2016-12-12:
nkii01@ubuntu:~/mbed/mbed-os-example-wifi$ git diff
diff --git i/main.cpp w/main.cpp
index 46cbac2..702f457 100644
--- i/main.cpp
+++ w/main.cpp
@@ -17,6 +17,8 @@
 #include "mbed.h"
 #include "TCPSocket.h"
 
+RawSerial output(USBTX, USBRX);
+
 #if TARGET_UBLOX_EVK_ODIN_W2
 #include "OdinWiFiInterface.h"
 OdinWiFiInterface wifi;
@@ -51,7 +53,7 @@ void scan_demo(WiFiInterface *wifi)
 {
     WiFiAccessPoint *ap;
 
-    printf("Scan:\r\n");
+    output.printf("Scan:\r\n");
 
     int count = wifi->scan(NULL,0);
 
@@ -62,11 +64,11 @@ void scan_demo(WiFiInterface *wifi)
     count = wifi->scan(ap, count);
     for (int i = 0; i < count; i++)
     {
-        printf("Network: %s secured: %s BSSID: %hhX:%hhX:%hhX:%hhx:%hhx:%hhx RSSI: %hhd Ch: %hhd\r\n", ap[i].get_ssid(),
+        output.printf("Network: %s secured: %s BSSID: %hhX:%hhX:%hhX:%hhx:%hhx:%hhx RSSI: %hhd Ch: %hhd\r\n", ap[i].get_ssid(),
                sec2str(ap[i].get_security()), ap[i].get_bssid()[0], ap[i].get_bssid()[1], ap[i].get_bssid()[2],
                ap[i].get_bssid()[3], ap[i].get_bssid()[4], ap[i].get_bssid()[5], ap[i].get_rssi(), ap[i].get_channel());
     }
-    printf("%d networks available.\r\n", count);
+    output.printf("%d networks available.\r\n", count);
 
     delete[] ap;
 }
@@ -75,7 +77,7 @@ void http_demo(NetworkInterface *net)
 {
     TCPSocket socket;
 
-    printf("Sending HTTP request to www.arm.com...\r\n");
+    output.printf("Sending HTTP request to www.arm.com...\r\n");
 
     // Open a socket on the network interface, and create a TCP connection to www.arm.com
     socket.open(net);
@@ -84,12 +86,12 @@ void http_demo(NetworkInterface *net)
     // Send a simple http request
     char sbuffer[] = "GET / HTTP/1.1\r\nHost: www.arm.com\r\n\r\n";
     int scount = socket.send(sbuffer, sizeof sbuffer);
-    printf("sent %d [%.*s]\r\n", scount, strstr(sbuffer, "\r\n")-sbuffer, sbuffer);
+    output.printf("sent %d [%.*s]\r\n", scount, strstr(sbuffer, "\r\n")-sbuffer, sbuffer);
 
     // Recieve a simple http response and print out the response line
     char rbuffer[64];
     int rcount = socket.recv(rbuffer, sizeof rbuffer);
-    printf("recv %d [%.*s]\r\n", rcount, strstr(rbuffer, "\r\n")-rbuffer, rbuffer);
+    output.printf("recv %d [%.*s]\r\n", rcount, strstr(rbuffer, "\r\n")-rbuffer, rbuffer);
 
     // Close the socket to return its memory and bring down the network interface
     socket.close();
@@ -97,27 +99,29 @@ void http_demo(NetworkInterface *net)
 
 int main()
 {
-    printf("WiFi example\r\n\r\n");
+    // Sets the console baud-rate
+    output.baud(115200);
+    output.printf("\r\nWiFi example\r\n\r\n");
 
     scan_demo(&wifi);
 
-    printf("\r\nConnecting...\r\n");
+    output.printf("\r\nConnecting...\r\n");
     int ret = wifi.connect(MBED_CONF_APP_WIFI_SSID, MBED_CONF_APP_WIFI_PASSWORD, NSAPI_SECURITY_WPA_WPA2);
     if (ret != 0) {
         printf("\r\nConnection error\r\n");
         return -1;
     }
 
-    printf("Success\r\n\r\n");
-    printf("MAC: %s\r\n", wifi.get_mac_address());
-    printf("IP: %s\r\n", wifi.get_ip_address());
-    printf("Netmask: %s\r\n", wifi.get_netmask());
-    printf("Gateway: %s\r\n", wifi.get_gateway());
-    printf("RSSI: %d\r\n\r\n", wifi.get_rssi());
+    output.printf("Success\r\n\r\n");
+    output.printf("MAC: %s\r\n", wifi.get_mac_address());
+    output.printf("IP: %s\r\n", wifi.get_ip_address());
+    output.printf("Netmask: %s\r\n", wifi.get_netmask());
+    output.printf("Gateway: %s\r\n", wifi.get_gateway());
+    output.printf("RSSI: %d\r\n\r\n", wifi.get_rssi());
 
     http_demo(&wifi);
 
     wifi.disconnect();
 
-    printf("\r\nDone\r\n");
+    output.printf("\r\nDone\r\n");
 }
diff --git i/mbed_app.json w/mbed_app.json
index 9adc47f..04d9575 100644
--- i/mbed_app.json
+++ w/mbed_app.json
@@ -2,11 +2,11 @@
     "config": {
         "wifi-ssid": {
             "help": "WiFi SSID",
-            "value": "\"SSID\""
+            "value": "\"4tousnet\""
         },
         "wifi-password": {
             "help": "WiFi Password",
-            "value": "\"Password\""
+            "value": "\"U!33p5k4rm\""
         }
     },
     "target_overrides": {

0xc0170, 2016-12-12:
cc @andreaslarssonublox

0xc0170, 2016-12-12:
@jankii01 what's that diff above about? The example should work out of the box, has been tested just recently with a release candidate.

I can see you are using RawSerial, does that fix it for you? Or what we should see in the diff above?

bulislaw, 2016-12-12:
Serial works for me just fine as it is and it worked for other OOB testers tried with multiple boards. Which board did you use?
We only support 2 WiFi modules atm both are supported in the example. Possibly we should use some better way of specifying pins for the ESP module.

jankii01, 2016-12-12:
UBLOX_EVK_ODIN_W2

jankii01, 2016-12-12:
Fix is to use RawSerial, set it's baudrate and excplictly put all printfs to it.

0xc0170, 2016-12-13:
> Fix is to use RawSerial, set it's baudrate and excplictly put all printfs to it.

should not be needed. Is this still valid , this issue

tommikas, 2017-01-25:
I also had to add Serial/RawSerial (either works) and set the baud rate for the prints to work.

Is the serial interface supposed to be initialized by default?

bulislaw, 2017-01-25:
It is initialized by default: https://github.com/ARMmbed/mbed-os/blob/master/platform/retarget.cpp#L110

tommikas, 2017-01-25:
Ah, of course. I was using the DAPLink USB serial port, which uses pins USBTX and USBRX instead of those.

andreaslarssonublox, 2017-01-26:
Can you have a look at this @andreaspeterssonublox?

bulislaw, 2017-02-06:
@tommikas can we close this issue?

tommikas, 2017-02-06:
As far as I'm concerned, sure. Up to @JanneKiiskila though.

bulislaw, 2017-02-15:
Closing it for now. Please reopen it if needed.

closed 2017-02-15 by bulislaw