ARMmbed/mbed-os-example-wifi #19: [OOB_5_4_RC1] uvision5 project generates link errors

toyowata, 2017-02-28:
When exported uvision5 project is compiled, I got following link errors.

```
linking...
.\BUILD\mbed-os-example-wifi.axf: Error: L6366E: atparser.o attributes are not compatible with the provided attributes .
Object atparser.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)

(snip)

Not enough information to list image symbols.
Not enough information to list the image map.
Finished: 23618 information, 0 warning and 5904 error messages.
".\BUILD\mbed-os-example-wifi.axf" - 5904 Error(s), 224 Warning(s).
Target not created.
Build Time Elapsed:  00:02:13

```

How to reproduce

```
$ mbed import https://github.com/ARMmbed/mbed-os-example-wifi
$ cd mbed-os-example-wifi/mbed-os
$ mbed update mbed-os-5.4.0-rc1
$ cd ..
$ mbed export -m NUCLEO_F429ZI -i uvision5
```

Open ```mbed-os-example-wifi.uvprojx``` and build the project.

bulislaw, 2017-02-28:
That looks like this issue https://github.com/ARMmbed/mbed-os/issues/3807
@theotherjimmy

0xc0170, 2017-03-02:
Will be fixed in RC2

closed 2017-03-03 by 0xc0170