ARMmbed/mbed-os-example-wifi #34: [OOB 5.5] HEXIWEAR + ESP8266 - CMIS-RTOS error

mray19027, 2017-06-08:
## Details

HEXIWEAR with ESP8266 module fails with the following logs. Additionally, had to make the following changes to the generic mbed-os-example-wifi repository:

mbed_app.json
```
"target_overrides": {
    ...
    "HEXIWEAR": {
        "wifi-tx": "PTD3",
        "wifi-rx": "PTD2"
    }
}
```

main.cpp
```
// #if TARGET_UBLOX_EVK_ODIN_W2
// #include "OdinWiFiInterface.h"
// OdinWiFiInterface wifi;

// #else
// #if !TARGET_FF_ARDUINO
// #error [NOT_SUPPORTED] Only Arduino form factor devices are supported at this time
// #endif
#include "ESP8266Interface.h"

ESP8266Interface wifi(MBED_CONF_APP_WIFI_TX, MBED_CONF_APP_WIFI_RX);

// #endif
```
Reasoning for the above change: The ESP8266 module I used was the HEXIWEAR ESP8266 module and therefore could run on a non-Arduino form factor.

## Debug details

GCC_ARM version: `5.4.1 20160919 (release)`
mbed CLI version: `1.0.0`
mbed OS version: `92fbf2a9b3988d430482fc25a6077f2462e2a634`

```
WiFi example

Scan:
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fe:80 RSSI: -67 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fe:81 RSSI: -66 Ch: 1
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:d1:80 RSSI: -83 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:d1:81 RSSI: -87 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:55:7d:c1 RSSI: -72 Ch: 1
Network: ESP_1CCF60 secured: None BSSID: 5E:CF:7F:1c:cf:60 RSSI: -46 Ch: 6
Network: GORDON-AP secured: None BSSID: B8:27:EB:d1:f9:c1 RSSI: -52 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:c0:61 RSSI: -69 Ch: 6
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -45 Ch: 6
Network: ESP_DC83BA secured: None BSSID: 1A:FE:34:dc:83:ba RSSI: -58 Ch: 6
Network: bdn_wireless secured: None BSSID: 8A:15:4:ca:90:f0 RSSI: -85 Ch: 6
Network: Brandywine Public secured: None BSSID: 8A:15:4:ca:90:f6 RSSI: -85 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:61:c0:60 RSSI: -69 Ch: 6
Network: AccessNG secured: Unknown BSSID: 70:3A:E:54:fc:60 RSSI: -81 Ch: 6
14 networks available.

Connecting...
Success

MAC: 18:fe:34:f9:e8:20
IP: 192.168.1.39
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: -49

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
CMSIS-RTOS error: Stack underflow (status: 0x1, task ID: 0x20001E00, task name: (null))
```

ashok-rao, 2017-06-09:
@bulislaw .. bump!

0xc0170, 2017-06-12:
cc @geky please have a look at this

kegilbert, 2017-06-12:
I saw this pop up on a K64F with GCC_ARM just now as well running this example. Reran and it did not show up again.

deepikabhavnani, 2017-06-14:
mbed OS version: 92fbf2a9b3988d430482fc25a6077f2462e2a634

While trying to reproduce above issue, I got new error:
''' Semaphore 20001ac8 error -17 '''
RtxSemaphore release reaches its max count.

adbridge, 2017-06-16:
@deepikabhavnani  Does #4560 actually fix this issue ?

adbridge, 2017-06-16:
Also if this an mbed-os issue rather than a specific issue with the example then this should be raised in mbed-os ?

ashok-rao, 2017-06-16:
@mray19027 , can you please move this issue to mbed OS repo? Looks like the mbed_app.json changes you made are platform specific so that can be picked up here and the other CMSIS error needs to go into mbed-os ..

deepikabhavnani, 2017-06-16:
@adbridge - #4560 resolves this issue. I will raise issue in mbed-os. How can i remove reference of PR from this issue?

adbridge, 2017-06-16:
@deepikabhavnani just delete the comment in your PR will do it :)

bulislaw, 2017-07-24:
Closing. Please reopen if it's still a problem.

closed 2017-07-25 by bulislaw