ARMmbed/mbed-os-example-wifi #35: [OOB 5.5.0] REALTEK_RTL8195AM cannot find any networks

mray19027, 2017-06-08:
REALTEK_RTL8195AM compiles properly, but cannot find any networks. 

Compile type: `GCC_ARM`
GCC ARM version: ` 5.4.1 20160919 (release)`
mbed OS: `92fbf2a9b3988d430482fc25a6077f2462e2a634`
mbed CLI version: `1.1.1`

Error log:
```
=========================================================

ROM Version: 0.3

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)

=========================================================
Check boot type form eFuse
SPI Initial
Image1 length: 0x2d64, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1
RTL8195A[HAL]: ISR 8 had been allocated!!w▒▒T-HY▒ample

Scan:
0 networks available.

Connecting...

Connection error
```

0xc0170, 2017-06-09:
cc @Archcady

Archcady, 2017-06-12:
I'm trying to reproduce this issue, but may I ask for more details? Thank you.

Archcady, 2017-06-12:
Compiler: GCC_ARM (gcc-arm-none-eabi-4_9-2015q3-win32)
mbed OS: 92fbf2a9b3988d430482fc25a6077f2462e2a634
running wifi example: https://github.com/ARMmbed/mbed-os-example-wifi/tree/ameba
with compile command: mbed compile -t GCC_ARM -m REALTEK_RTL8195AM --app-config .\mbed-os-example-wifi-ameba\mbed_app.json -vv
under mbed-cli version 1.1.1
log
```

<RTL8195A>
=========================================================

ROM Version: 0.2

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003) 

=========================================================

Check boot type form eFuse

SPI Initial

Image1 length: 0x2d64, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

Initializing WIFI ...

WIFI initialized

WIFI is already runningWiFi example

Scan:
Network: iot_dlink secured: WPA/WPA2 BSSID: 78:54:2E:4e:19:80 RSSI: -36 Ch: 1
Network: iot_sd2 secured: WPA2 BSSID: 0:E0:4C:81:97:44 RSSI: -38 Ch: 1
Network: ftp_test3 secured: None BSSID: 0:E0:4C:81:96:d1 RSSI: -38 Ch: 11
Network: Singtel8900-D728 secured: WPA/WPA2 BSSID: E0:8E:3C:26:d7:29 RSSI: -42 Ch: 2
Network: iot_simpleConfig secured: WPA/WPA2 BSSID: C4:6E:1F:58:42:bc RSSI: -44 Ch: 1
Network: RealEZ secured: WPA2 BSSID: 9C:1C:12:13:cd:1 RSSI: -44 Ch: 1
Network: Remote-EZ secured: WPA2 BSSID: 80:2A:A8:44:ac:11 RSSI: -44 Ch: 6
Network: RealKungFu secured: WPA2 BSSID: 9C:1C:12:13:cd:0 RSSI: -46 Ch: 1
Network: Zhone_8225 secured: WPA2 BSSID: 72:2:71:32:82:26 RSSI: -60 Ch: 2
Network:  secured: None BSSID: 0:E0:4C:72:72:72 RSSI: -62 Ch: 11
Network: RealKungFu secured: WPA2 BSSID: 24:DE:C6:ba:bd:d1 RSSI: -64 Ch: 1
Network: Remote-KungFu secured: WPA2 BSSID: 82:2A:A8:44:ac:11 RSSI: -70 Ch: 6
Network: ZT-SG-Mobile secured: WPA/WPA2 BSSID: 54:B8:A:2c:34:b0 RSSI: -70 Ch: 6
Network:  secured: None BSSID: 0:E0:4C:14:cc:58 RSSI: -74 Ch: 11
14 networks available.

Connecting...

RTL8195A[Driver]: set ssid [RealEZ] 

RTL8195A[Driver]: start auth to 9c:1c:12:13:cd:01

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=2)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
```

Archcady, 2017-06-12:
And this is a test on our ROM version 0.3 board, which should be the same board just like yours. 
I changed the WiFi security to NSAPI_SECURITY_WPA2.
```
<RTL8195A>
=========================================================

ROM Version: 0.3

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003) 

=========================================================

Check boot type form eFuse

SPI Initial

Image1 length: 0x2d64, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

Initializing WIFI ...

WIFI initialized

WIFI is already runningWiFi example

Scan:
Network: ftp_test3 secured: None BSSID: 0:E0:4C:81:96:d1 RSSI: -76 Ch: 11
Network: iot_sd2 secured: WPA2 BSSID: 0:E0:4C:81:97:44 RSSI: -80 Ch: 1
Network: RealKungFu secured: WPA2 BSSID: 9C:1C:12:13:cd:0 RSSI: -82 Ch: 1
Network: iot_dlink secured: WPA/WPA2 BSSID: 78:54:2E:4e:19:80 RSSI: -82 Ch: 1
Network: RealEZ secured: WPA2 BSSID: 9C:1C:12:13:cd:1 RSSI: -84 Ch: 1
Network: Singtel8900-D728 secured: WPA/WPA2 BSSID: E0:8E:3C:26:d7:29 RSSI: -86 Ch: 1
Network: Remote-KungFu secured: WPA2 BSSID: 82:2A:A8:44:ac:11 RSSI: -86 Ch: 6
Network: Remote-EZ secured: WPA2 BSSID: 80:2A:A8:44:ac:11 RSSI: -86 Ch: 6
Network: iot_simpleConfig secured: WPA/WPA2 BSSID: C4:6E:1F:58:42:bc RSSI: -88 Ch: 1
Network: Zhone_8225 secured: WPA2 BSSID: 72:2:71:32:82:26 RSSI: -90 Ch: 2
10 networks available.

Connecting...

RTL8195A[Driver]: set ssid [RealEZ] 

RTL8195A[Driver]: start auth to 9c:1c:12:13:cd:01

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=2)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1
Success

MAC: 00:e0:4c:e2:e2:e2
IP: 172.25.24.19
Netmask: 255.255.255.0
Gateway: 172.25.24.253
RSSI: -85

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

ioctl[SIOCGIWESSID] ssid = NULL, not connected
Done
```

0xc0170, 2017-06-12:
@Archcady Might not change things, but please switch to gcc 6 (mbed OS 5.5 and upwards)

Archcady, 2017-06-12:
I test again with GCC 6 and using exactly the wifi example, nothing changed. Here's the log. 
```
<RTL8195A>
=========================================================

ROM Version: 0.3

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003) 

=========================================================

Check boot type form eFuse

SPI Initial

Image1 length: 0x2d64, Image Addr: 0x10000bc8

Image1 Validate OK, Going jump to Image1

Initializing WIFI ...

WIFI initialized

WIFI is already runningWiFi example

Scan:
Network: iot_sd2 secured: WPA2 BSSID: 0:E0:4C:81:97:44 RSSI: -40 Ch: 1
Network: Singtel8900-D728 secured: WPA/WPA2 BSSID: E0:8E:3C:26:d7:29 RSSI: -40 Ch: 1
Network: iot_dlink secured: WPA/WPA2 BSSID: 78:54:2E:4e:19:80 RSSI: -44 Ch: 1
Network: Remote-EZ secured: WPA2 BSSID: 80:2A:A8:44:ac:11 RSSI: -44 Ch: 6
Network: Remote-KungFu secured: WPA2 BSSID: 82:2A:A8:44:ac:11 RSSI: -44 Ch: 6
Network: iot_simpleConfig secured: WPA/WPA2 BSSID: C4:6E:1F:58:42:bc RSSI: -46 Ch: 1
Network: RealEZ secured: WPA2 BSSID: 9C:1C:12:13:cd:1 RSSI: -46 Ch: 1
Network: RealKungFu secured: WPA2 BSSID: 9C:1C:12:13:cd:0 RSSI: -48 Ch: 1
Network: AmebaTest secured: WPA/WPA2 BSSID: A0:AB:1B:13:1b:6f RSSI: -48 Ch: 11
Network: Zhone_8225 secured: WPA2 BSSID: 72:2:71:32:82:26 RSSI: -56 Ch: 2
Network:  secured: None BSSID: 0:E0:4C:14:cc:58 RSSI: -56 Ch: 11
Network: RealKungFu secured: WPA2 BSSID: 24:DE:C6:ba:bd:d1 RSSI: -60 Ch: 1
Network: RealEZ secured: WPA2 BSSID: 24:DE:C6:ba:bd:d0 RSSI: -62 Ch: 1
Network: ZT-SG-Mobile secured: WPA/WPA2 BSSID: 54:B8:A:2c:38:0 RSSI: -62 Ch: 11
Network: ZT-SG-Mobile secured: WPA/WPA2 BSSID: 54:B8:A:2c:34:b0 RSSI: -68 Ch: 6
15 networks available.

Connecting...

RTL8195A[Driver]: set ssid [RealEZ] 

RTL8195A[Driver]: start auth to 9c:1c:12:13:cd:01

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=1)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:1

Success

MAC: 00:e0:4c:e2:e2:e2
IP: 172.25.24.19
Netmask: 255.255.255.0
Gateway: 172.25.24.253
RSSI: -48

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

ioctl[SIOCGIWESSID] ssid = NULL, not connected
Done
```

kegilbert, 2017-06-12:
Running with GCC (mbed compile -t GCC_ARM -m REALTEK_RTL8195AM) on the iotlab network and the ameba branch example [https://github.com/ARMmbed/mbed-os-example-wifi/tree/ameba] I get

```
=========================================================

ROM Version: 0.3

Build ToolChain Version: gcc version 4.8.3 (Realtek ASDK-4.8.3p1 Build 2003)

=========================================================
Check boot type form eFuse
SPI Initial
Image1 length: 0x2d64, Image Addr: 0x10000bc8
Image1 Validate OK, Going jump to Image1

Initializing WIFI ...
WIFI initialized

WIFI is already runningWiFi example

Scan:
Network: ESP_DCDD7D secured: None BSSID: 1A:FE:34:dc:dd:7d RSSI: -36 Ch: 10
Network: AccessNG secured: WPA2 BSSID: 94:B4:F:4a:37:80 RSSI: -36 Ch: 11
Network: IoTBU-Sniffer secured: WPA/WPA2 BSSID: 20:C:C8:4b:2a:ed RSSI: -38 Ch: 10
Network: Guest-AccessNG secured: None BSSID: 94:B4:F:4a:37:81 RSSI: -38 Ch: 11
Network: iotlab secured: WPA2 BSSID: E4:F4:C6:12:32:9a RSSI: -44 Ch: 11
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:54:fe:80 RSSI: -52 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fe:81 RSSI: -54 Ch: 1
Network: mbed secured: WPA2 BSSID: 18:D6:C7:52:87:dc RSSI: -56 Ch: 11
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:61:c0:60 RSSI: -58 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:61:c0:61 RSSI: -62 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:55:7d:c0 RSSI: -68 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:55:7d:c1 RSSI: -68 Ch: 1
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:54:fc:60 RSSI: -68 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:54:fc:61 RSSI: -68 Ch: 6
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:61:c9:20 RSSI: -72 Ch: 11
15 networks available.

Connecting...

RTL8195A[Driver]: set ssid [iotlab]

RTL8195A[Driver]: start auth to e4:f4:c6:12:32:9a

RTL8195A[Driver]: auth success, start assoc

RTL8195A[Driver]: association success(res=2)

RTL8195A[Driver]: set pairwise key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4)

RTL8195A[Driver]: set group key to hw: alg:4(WEP40-1 WEP104-5 TKIP-2 AES-4) keyid:2
```

Archcady, 2017-06-13:
Yes, we expect that the wifi example will hang before it connects to wifi due to lwip getting initialized twice. Right now we are discuss the possible solution to this issue, see [here](https://github.com/ARMmbed/mbed-os/pull/4444)

c1728p9, 2017-06-27:
Hi @Archcady, did https://github.com/ARMmbed/mbed-os/pull/4444 fix the issue?

Archcady, 2017-06-28:
Hi @c1728p9 , We can not confirm the original issue reported by @mray19027 is fixed because we are unable to reproduce. But another issue is fixed that our lwip no longer init twice so the program is no longer hang at connect part.

0xc0170, 2017-06-28:
@mray19027 please confirm

bulislaw, 2017-07-24:
Closing. Please reopen if it's still a problem.

closed 2017-07-25 by bulislaw