ARMmbed/mbed-os-example-wifi #36: [OOB 5.5.0] ublox_evk_odin_w2 never gets to main

mray19027, 2017-06-09:
ublox_evk_odin_w2 never gets to main. Compiles just fine, deploy to the device, but then hangs and never displays anything or does anything.

Compiler: IAR, GCC_ARM, and ARM
GCC_ARM version: `5.4.1 20160919 (release)`
mbed CLI version: `1.1.1`
mbed OS version: `92fbf2a9b3988d430482fc25a6077f2462e2a634`

0xc0170, 2017-06-12:
cc @andreaslarssonublox

ashok-rao, 2017-06-12:
@mray19027  may not change anything major, but can you please try gcc v6 (mbed OS 5.5 officially supported)..instructions are in OoB spreadsheet..

andreaslarssonublox, 2017-06-12:
We'll have a look at it.
int ref TE_OM2-72

tomasfrisbergublox, 2017-06-14:
@0xc0170 

It seems to be two separate problems. One of them requires a new driver release from us. This will be done within the next couple of days.

Also, it seems to be a problem in how the interfaces are registered into the lwip stack.

In “lwip_stack.c”, the function “mbed_lwip_bringup_2” calls “mbed_lwip_emac_init(NULL)”, which calls the “netif_add” function in “lwip_netif.c”.

The NULL parameter will be a problem since it is forwarded as the state variable to the “netif_add” function.
  /* remember netif specific state information data */
  netif->state = state;

Further down, in “netif_add”,  the installed “init” callback generates a call to “emac_lwip_if_init” in “emac_lwip.c”. Here the state variable is cast to the emac pointer which ends up in a null pointer crash.
err_t emac_lwip_if_init(struct netif *netif)
{
    int err = ERR_OK;
    emac_interface_t *mac = (emac_interface_t *)netif->state;

    mac->ops.set_link_input_cb(mac, emac_lwip_input, netif);
    mac->ops.set_link_state_cb(mac, emac_lwip_state_change, netif);

Hence, according to my understanding this cannot work for the DEVICE_EMAC solution.

The solution could be either of the two proposals
1.	An “mbed_lwip_emac_initiated” variables that makes sure the code of the “mbed_lwip_emac_init” function is executed only once. Since we do a separate “mbed_lwip_emac_init” call when registering the interface this means it will not execute code from the bringup2 function.
2.	Replace the “mbed_lwip_emac_init(NULL)” with “ret = NSAPI_ERROR_OK;” since it will not work for the DEVICE_EMAC case.
#if !DEVICE_EMAC
            ret = mbed_lwip_emac_init(NULL);
#else
            ret = NSAPI_ERROR_OK;
#endif

0xc0170, 2017-06-14:
@tomasfrisbergublox Thanks for the description. This seems is the same issue as this patch is fixing : https://github.com/ARMmbed/mbed-os/pull/4444 ? Please review 

In case it is not, this would be worth having it in mbed-os repository, can you please add it there?

tomasfrisbergublox, 2017-06-14:
@0xc0170 

The ARMmbed/mbed-os#4444 patch is working also for us. We will continue testing and generate a new binary hopefully tomorrow.

Thanks

0xc0170, 2017-06-14:
> The ARMmbed/mbed-os#4444 patch is working also for us. We will continue testing and generate a new binary hopefully tomorrow.

Please do, so it can be added to 5.5.1

0xc0170, 2017-06-16:
The status update, this issue has some dependencies

- https://github.com/ARMmbed/mbed-os/pull/4543(5.5.0)
- https://github.com/ARMmbed/mbed-os/pull/4444 (5.5.1)
- https://github.com/ARMmbed/mbed-os/pull/4567 (5.5.1/2)

Once the last one gets integrated, it should work . @tomasfrisbergublox please correct me if it is not the case

MarceloSalazar, 2017-06-28:
As 5.5.1 is released, can @tomasfrisbergublox please confirm whether this is fixed and can be closed?

bulislaw, 2017-07-24:
Closing. Please reopen if it's still a problem.

closed 2017-07-24 by bulislaw