ARMmbed/mbed-os-example-wifi #54: ODIN support

andcor02, 2017-09-12:
Hi All,

I am aware this example supports the ODIN, but a number of people have flagged how do you actually build the example for the ODIN?? The instructions are very unclear, and they just state using a shield.

Please sort this....

Regards,

Andrea

janjongboom, 2017-09-12:
@bulislaw @adbridge

adbridge, 2017-09-12:
@bulislaw over to you :)

bulislaw, 2017-09-12:
It's actually explicitly mentioned in the README:

> Compile and generate binary
> For example, for GCC:
> mbed compile -t GCC_ARM -m UBLOX_EVK_ODIN_W2

bulislaw, 2017-09-12:
Improvements to docs are welcomed :)

janjongboom, 2017-09-12:
@bulislaw Yes, I'm aware, but it's unclear. If we can split this up in two sections, one for boards with built-in wifi and one for using shields, that'd help.

yennster, 2017-09-15:
Also, for boards using built-in wifi, should the instructions also state whether to include the following in `.mbedignore`?:

```
wifi-x-nucleo-idw01m1/*
esp8266-driver/*
```

janjongboom, 2017-09-27:
@SeppoTakalo Why are we closing this?! The documentation is unclear and it should be fixed. This issue should be reopened.

SeppoTakalo, 2017-09-27:
Ublox is mentioned in https://github.com/ARMmbed/mbed-os-example-wifi#supported-hardware

Also in the [Getting started](https://github.com/ARMmbed/mbed-os-example-wifi#getting-started)
 **Configure the WiFi shield to use** section have this clause:
```
For built-in WiFi, ignore the value of wifi-shield
```

And next section `For built-in WiFi, keep the default .mbedignore`

Is this not enough?

Each step have built-in wifi mentioned?

janjongboom, 2017-09-27:
@andcor02 What's your opinion?

andcor02, 2017-09-28:
I think this is now fine, its a lot more clear.

Regards

Andrea Corrado
Graduate Engineer
ARM
andrea.corrado@arm.com
Skype: andrea.corrado.arm
Mobile: + 44 7901333702

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.