ARMmbed/mbed-os-example-wifi #58: [Odin

yennster, 2017-09-15:


yennster, 2017-09-15:
Closed since I may have an issue with VM, investigating.

yennster, 2017-09-15:
Investigated issue, still occurs on my native Mac environment as well with GCC_ARM:

```
$$$.12360221022268373D66EB3C
```

adbridge, 2017-09-18:
Seppo's team is taking a look

SeppoTakalo, 2017-09-19:
I can verify that ODIN W2 with following code can produce the same behaviour:
```
#include "mbed.h"
#include "OdinWiFiInterface.h"

int main()
{
    printf("Initialising OdinWiFiInterface\n");
    OdinWiFiInterface wifi;

    printf("Done\n");
    return 0;
}
```

The `Initialising OdinWiFiInterface` is printed but `Done` is not so assuming that board crashes.

SeppoTakalo, 2017-09-19:
Tested the above code with different mbed OS versions
|mbed-os tag|works?|
|-----------|------|
|5.6-rc1|no|
|5.5.7|no|
|5.5.0|no|
|5.4.0| does not compile with mbed Cli 1.2.0 |

I cannot find a revision where it does not crash.
I wonder has there been some kind of firmware change that breaks the compatibility.

SeppoTakalo, 2017-09-19:
I modified the `main.cpp` to following and removed the `mbed_app.json`

```
#include "mbed.h"
#include "OdinWiFiInterface.h"

int main()
{
    printf("Initialising OdinWiFiInterface\r\n");
    wait(0.5);
    OdinWiFiInterface wifi;

    printf("Done\r\n");
    return 0;
}
```

This is the simplest test case possible for the Odin. All settings in default, no code changes.

Now replying the testcase for range of Mbed OS revisions

| tag/branch | works |
|--------|-------|
| master | no |
| mbed-os-5.6.0-rc1 | no |
| mbed-os-5.5.7 | no |
| mbed-os-5.5.1 | yes |
| mbed-os-5.5.0 | no |
| mbed-os-5.4.7 | yes |
| mbed-os-5.4.5 | yes |

This seems to have been known problem in Mbed OS 5.5 release time, but never reported
in the known issues list. Fix seems to have been merged in 5.5.1, but re-appears later.

* https://github.com/ARMmbed/mbed-os-example-wifi/issues/36
* https://github.com/ARMmbed/mbed-os/issues/4663
* https://github.com/ARMmbed/mbed-os/pull/4567

SeppoTakalo, 2017-09-19:
Reported in https://github.com/ARMmbed/mbed-os/issues/5139

screamerbg, 2017-09-19:
cc @andreaslarssonublox @andreaspeterssonublox

andreaslarssonublox, 2017-09-19:
Internal u-blox jira ticket TE_OM2-97

SeppoTakalo, 2017-10-09:
Seems to be fixed in Mbed OS 5.6.2

closed 2017-10-09 by SeppoTakalo