ARMmbed/mbed-os-example-wifi #60: [OOB 5.6] Incomplete Realtek RTL8195AM Instructions

dlfryar, 2017-09-19:
No library included for RTL8195AM board to use with WiFi example.  Instructions mention only two modules (esp8266 and idw01m1).

[REALTEK-RTL8195AM](https://os.mbed.com/platforms/REALTEK-RTL8195AM) page does not give pinout information for mbed_app.json

```
mbed target
[mbed] REALTEK_RTL8195AM
```

```
mbed toolchain
[mbed] arm
```

Compiling results in a error around the ATParser due to no library

```
mbed compile
Building project mbed-os-example-wifi (REALTEK_RTL8195AM, ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [ 97.8%]: BufferedSerial.cpp
[Warning] BufferedSerial.cpp@31,0:  #1361-D: function "mbed::SerialBase::attach(T *, void (T::*)(), mbed::SerialBase::IrqType) [with T=BufferedSerial]"  (declared at line 117 of "./mbed-os/drivers/SerialBase.h") was declared "deprecated"
[Warning] BufferedSerial.cpp@156,0:  #1361-D: function "mbed::SerialBase::attach(T *, void (T::*)(), mbed::SerialBase::IrqType) [with T=BufferedSerial]"  (declared at line 117 of "./mbed-os/drivers/SerialBase.h") was declared "deprecated"
Compile [ 98.1%]: ATParser.cpp
Compile [ 98.4%]: ESP8266.cpp
Compile [ 98.8%]: ESP8266Interface.cpp
Compile [ 99.1%]: BufferedSerial.cpp
Compile [ 99.4%]: ATParser.cpp
Compile [ 99.7%]: SPWFSA01.cpp
[Error] SPWFSA01.cpp@26,0:  #289: no instance of constructor "ATParser::ATParser"  matches the argument list
[Error] SPWFSA01.cpp@27,0:  #20: identifier "PC_12" is undefined
[ERROR] "./wifi-x-nucleo-idw01m1/SPWF01SA/SPWFSA01.cpp", line 26: Error:  #289: no instance of constructor "ATParser::ATParser"  matches the argument list
            argument types are: (BufferedSerial, const char [2], const char [2])
"./wifi-x-nucleo-idw01m1/SPWF01SA/SPWFSA01.cpp", line 27: Error:  #20: identifier "PC_12" is undefined
./wifi-x-nucleo-idw01m1/SPWF01SA/SPWFSA01.cpp: 0 warnings, 2 errors

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/ubuntu/Downloads/mbed-os-example-wifi/mbed-os/tools/make.py -t arm -m REALTEK_RTL8195AM --source . --build ./BUILD/REALTEK_RTL8195AM/arm" in "/home/ubuntu/Downloads/mbed-os-example-wifi"
```

dlfryar, 2017-09-19:
@bulislaw @SeppoTakalo is there another location for the mbed_app.json config and libraries for this board/example?

MarceloSalazar, 2017-09-19:
Note, I've been able to compile for the REALTEK_RTL8195AM by using `.mbedignore` (including both `esp8266-driver/*` and `wifi-x-nucleo-idw01m1/*`).
@dlfryar could you give it a try and confirm whether it works?

IMO, configuring this example is becoming a bit ugly (having to use mbedignore files + mbed_app.json)
Would it make sense to use `mbed add xxx` (whether xxx is the corresponding wifi driver?)
This won't be required for platforms with native wifi support (odin-w2, rtls8195).

@bulislaw any thoughts on this?

dlfryar, 2017-09-19:
Both drivers in .mbedignore does compile.  I agree that some work needs to be done with either the documentation or process.  The way it compiles now is not intuitive.  Looking at mbed_app.json shows a module configuration but .mbedignore filters that out.

cat .mbedignore

```
wifi-x-nucleo-idw01m1/*
esp8266-driver/*
```

dlfryar, 2017-09-20:
Here's a copy of **`mbed_app.json`** and **`.mbedignore`** that this example needs to build on the Realtek RTL8195AM.

**_mbed_app.json_**
```
{
  "config": {
    "wifi-ssid": {
      "help": "WiFi SSID",
      "value": "\"my_wifi_ssid\""
    },
    "wifi-password": {
      "help": "WiFi Password",
      "value": "\"my_wifi_password\""
    }
  }
}
```

**_.mbedignore_**

```
wifi-x-nucleo-idw01m1/*
esp8266-driver/*

```

JanneKiiskila, 2017-09-25:
What would really help this issue would be to either
a) get everything compile nicely w/o issues (@sg-) OR
b) get a way to include the `.mbedignore` file in the `mbed_app.json` , as you can point a specific `mbed_app.json` for `mbed compile`, but you can't give it a `.mbedignore`.

sg-, 2017-09-25:
The pin definition can be fixed easily but the duplication of ATParser is the bigger problem. These should both be updated to use the version in Mbed OS. @SenRamakri

closed 2017-09-26 by SeppoTakalo

SeppoTakalo, 2017-09-26:
Duplicated ATParser can be removed by clicking a merge button in the driver.
Just give the ownership to me and I can fix it.

PR #63 (2017-09-26): Fix OOB issues
* Document and fix the build when using on-board WiFi
* Remove form factor checks to allow K64F (of other) with ESP8266

Fixes #60, #61 and #54
commit 795910ef (2017-09-26): Add default .mbedignore file

By default, we use built-in drivers and require user to replace
.mbedignore file when using one of the provided Wifi shield.

Fixes #60