ARMmbed/mbed-os-example-wifi #90: Re-compile the mbed-os-example-wifi with ARM Compiler 5, gives error on REALTEK_RTL8195AMboard

jamesbeyond, 2018-03-07:
REALTEK_RTL8195AM is a support board for this example.

I can compile the example for REALTEK_RTL8195AM board, with ARM Compiler 5 by the command:
`mbed compile -t ARM -m REALTEK_RTL8195AM`
The example, will built without errors.

but, if I change some of the code (e.g. change the Wifi password in mbed_app.json ). and re-compile the example again:
`mbed compile -t ARM -m REALTEK_RTL8195AM`

Error throw:
```
Elf2Bin: mbed-os-example-wifi
Error: Q0147E: Failed to create Directory .\BUILD\REALTEK_RTL8195AM\ARM\mbed-os-example-wifi.bin\IMAGE2_TABLE: File exists
Finished: 0 information, 0 warning and 1 error messages.
[ERROR] Error: Q0147E: Failed to create Directory .\BUILD\REALTEK_RTL8195AM\ARM\mbed-os-example-wifi.bin\IMAGE2_TABLE: File exists
Finished: 0 information, 0 warning and 1 error messages.
```

This could be a tools problem.
But the problem only happens on Re-compile. if I do a clean first, the compiling would be fine.
Only with  REALTEK_RTL8195AM boar, other boards are fine.
Only with ARM Compiler 5. Gcc and IAR are fine.

theotherjimmy, 2018-03-08:
> This could be a tools problem.

This looks like a limitation of ARM compiler 5. I'm not sure what to do about it.

jamesbeyond, 2018-03-08:
Raised a new issue on mbed-OS:
https://github.com/ARMmbed/mbed-os/issues/6306
So closing this one

closed 2018-03-08 by jamesbeyond