ARMmbed/mbed-os-example-wifi #91: Compiling the example with online IDE for realtek_rtl8195am board throw out errors.

jamesbeyond, 2018-03-07:
I am using mbed online IDE to compile the example for realtek_rtl8195am board. online IDE throw compiling errors:

Steps:
1. Open mbed IDE, import `mbed-os-example-wifi example`
2. select `realtek_rtl8195am` platform.
3. compiling.

[image]

theotherjimmy, 2018-03-08:
Yeah, you're not using the RC.

jamesbeyond, 2018-03-08:
@theotherjimmy You mean to update the example with 5.8.0-rc branch? I did that after import the example, however it is the same result. I can update the summary to reflect that.

BTW, I only update to the RC in the example level, however, the mbed-os within the example seems not got updated with the example from the online compiler. This was mentioned in the Slack threads by Seppo Takalo

theotherjimmy, 2018-03-08:
No, I mean update mbed-os. That's a bug in 5.7.6, I think.

SeppoTakalo, 2018-06-04:
Not visible anymore, so fixed somewhere in Mbed OS 5.8.

closed 2018-06-04 by SeppoTakalo