ARMmbed/mbed-os-example-wifi #92: The mbed-os-example-wifi compiled by mbed online IDE not working on UBLOX_EVK_ODIN_W2

jamesbeyond, 2018-03-07:
I am using mbed online IDE to compile the example for UBLOX_EVK_ODIN_W2 board. online IDE can compile the board, and generate a binary image. But the binary image is not working on the board.

However, if I just use mbed cli to compile the same example with arm compiler 5, the example works fine on my UBLOX_EVK_ODIN_W2 board.

SeppoTakalo, 2018-03-08:
It is because online compiler uses "latest" mbed OS release, which is 5.7.6.
Not 5.8.0-rc1

This issue is same as:
https://github.com/ARMmbed/mbed-os/issues/5139

I believe you can replicate the result by trying to compile against 5.7.6 on command line.

Closing as dublicate.

closed 2018-03-08 by SeppoTakalo

jamesbeyond, 2018-03-08:
Firstly, the online IDE, I was using the dev version from https://os-partnerbeta.mbed.com/
Secondly, the wifi example I imported, I manually changed the branch of example to 5.8.0-oob.
In my environment, seems everything been updated to 5.8.0 rc, so I am not sure if this problem caused by 5.7.6

SeppoTakalo, 2018-03-08:
Checking application branch `5.8.0-rc1` from the application does not seem to check out the `5.8.0-rc1` from mbed-os.

Lets investigate this on the Slack channel...