ARMmbed/mbed-os-example-wifi #94: Running this on NUCLEO_L476RG + ST Wifi X-NUCLEO-IDW01A1 prints error _recv_ap (212) on serial port

GarySwansonARM, 2018-03-19:
On windows.
mbed import mbed-os-example-wifi
mbed update mbed-os-5.8.0-oob
mbed compile -t IAR -m NUCLEO_L476RG
flash to board & reset
On the serial port I see:
Scan:
_recv_ap (212) - ERROR: Should never happen!
Network: Hidden-Network secured: WPA2 BSSID: F8:B:CB:43:46:80 RSSI: -83 Ch: 1
Network: AccessNG secured: WPA2 BSSID: 70:3A:E:64:4e:c0 RSSI: -84 Ch: 1
Network: TP-LINK_ACD326 secured: WPA2 BSSID: C4:E9:84:ac:d3:26 RSSI: -69 Ch: 3
Network: Guest-AccessNG secured: None BSSID: 70:3A:E:64:95:c1 RSSI: -57 Ch: 6
Network: ASUS secured: Unknown BSSID: 1C:B7:2C:7c:3d:c0 RSSI: -53 Ch: 6
5 networks available.

GarySwansonARM, 2018-03-19:
This happened for about 5 of the reboots, and has now not happened for the past 3.

SeppoTakalo, 2018-03-20:
Obviously a parser error in WiFi driver: https://github.com/ARMmbed/wifi-x-nucleo-idw01m1/blob/5871f7011d7ff2c50e3faf992ebab88e9f69dc95/SPWFSA01/SPWFSA01.cpp#L212

I cannot replicate that same with different board, new version of driver and IAR.

Can you try:
```
cd wifi-x-nucleo-idw01m1
git checkout master
git pull
cd ..
```
And recompile & retest.

The driver seems to be very old and probably needs updating.

SeppoTakalo, 2018-06-04:
No updates, closing as the root cause was most probably outdated driver.

closed 2018-06-04 by SeppoTakalo