ARMmbed/mbed-os-example-wifi #95: K66F with ESP8266 connection error

cargro, 2018-03-20:
using mbed-os-5.8-oob branch

mbed_app.json file:

[image]

serial output: 
[image]

SeppoTakalo, 2018-03-20:
Did you re-try it?
Connecting to WiFi AP is not 100% success every time.

Also, you blacked you the WiFi AP list so I cannot very whether the target was even in range.

Anyway, I cannot replicate this in our internal CI. I can provide you a link on separate email.

ashok-rao, 2018-03-20:
@SeppoTakalo .. I confirm that the AP was in range and we did re-try a few times.. we are using the ESP8266 + K66F.. we also redefined the wifi-tx/rx pins in mbed_app.json just to be 100% sure it was not something related to pins.. we also tried using a mobile phone as hotspot (with respective SSID/pwd).. but even that did not connect..

SeppoTakalo, 2018-06-04:
I'm closing this as I'm not seeing the same issues in our CI:
https://jenkins-internal.mbed.com/view/IP%20Core/job/mbed-os-wifi-test/compiler=ARM,protocol=IPv4,target=K64F,variant=ESP8266/HTML_20Report/

closed 2018-06-04 by SeppoTakalo