ARMmbed/mbed-os-example-wifi #97: "mbed-os-example-wifi " can't start ODIN-W2 with 5.8.0(GCC)

okoshi3, 2018-03-28:
I was expecting that it will be resolved in 5.8.0 with the message in 5.7.6.
However, it did not move after all. It becomes the following message.

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : FFFFFFFF
R1   : 200020F0
R2   : 000009F6
R3   : 0803D5E7
R4   : 00000000
R5   : 00000000
R6   : 53002400
R7   : 34510100
R8   : 30363234
R9   : 200114F4
R10  : 00000000
R11  : 00000008
R12  : 20010C64
SP   : 20010D44
LR   : 000178C1
PC   : 080155B2
xPSR : 41000200
PSP  : 20010D20
MSP  : 2002FFD8
CPUID: 410FC241
HFSR : 80000000
MMFSR: 00000000
BFSR : 00000000
UFSR : 00000000
DFSR : 0000000A
AFSR : 00000000
SHCSR: 00000000

Thread Info:
Current:
State: 00000002 EntryFn: 0800E501 Stack Size: 00001000 Mem: 2000FE90 SP: 20010BD0
Next:
State: 00000002 EntryFn: 0800E501 Stack Size: 00001000 Mem: 2000FE90 SP: 20010BD0
Wait Threads:
State: 00000083 EntryFn: 0801144D Stack Size: 00000300 Mem: 20001B10 SP: 20001DA8
Delay Threads:
State: 00000043 EntryFn: 08011519 Stack Size: 00000200 Mem: 20012490 SP: 200125F8
Idle Thread:
State: 00000001 EntryFn: 0800E689 Stack Size: 00000200 Mem: 20001E10 SP: 20001FD0

-- MbedOS Fault Handler --

Compile Mesage
C:\L\src\mbed\tmp\mbed-os-example-wifi>mbed compile -c -m UBLOX_EVK_ODIN_W2 -t GCC_ARM
Building project mbed-os-example-wifi (UBLOX_EVK_ODIN_W2, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [  0.2%]: AnalogIn.cpp
Compile [  0.4%]: main.cpp
[Warning] MbedCRC.h@34,32: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]

[Warning] MbedCRC.h@34,32: unknown option after '#pragma GCC diagnos
Link: mbed-os-example-wifi
Elf2Bin: mbed-os-example-wifi
+-----------------------------------+---------+-------+-------+
| Module                            |   .text | .data |  .bss |
+-----------------------------------+---------+-------+-------+
| [fill]                            |     402 |    13 |    70 |
| [lib]\c.a                         |   39641 |  2212 |    56 |
| [lib]\gcc.a                       |    3976 |     0 |     0 |
| [lib]\m.a                         |      88 |     0 |     0 |
| [lib]\misc                        |     236 |    16 |    28 |
| [lib]\ublox-odin-w2-driver.a      |  927264 |    88 | 12499 |
| esp8266-driver\ESP8266Interface.o |      16 |     0 |     0 |
| main.o                            |    1337 |     4 |   864 |
| mbed-os\drivers                   |     772 |     0 |     0 |
| mbed-os\events                    |    1458 |     0 |   101 |
| mbed-os\features                  |   53018 |   170 | 46934 |
| mbed-os\hal                       |    1510 |     4 |    66 |
| mbed-os\platform                  |    2307 |   260 |   149 |
| mbed-os\rtos                      |   15940 |   168 |  6065 |
| mbed-os\targets                   |   17163 |     5 |   680 |
| Subtotals                         | 1065128 |  2940 | 67512 |
+-----------------------------------+---------+-------+-------+
Total Static RAM memory (data + bss): 70452 bytes
Total Flash memory (text + data): 1068068 bytes

mikaleppanen, 2018-03-28:
Hi,  I cannot repeat the error here with UBLOX_EVK_ODIN_W2. What version of the gcc are you using? I am seeing a bit different memory sizes in my local build than those above.

okoshi3, 2018-03-28:
GCC ver is 4.9 2015q3
---------------------------
C:\L\src\mbed\tmp\mbed-os-example-wifi>mbed compile -c -v -m UBLOX_EVK_ODIN_W2 -t GCC_ARM
[mbed] Working path "C:\L\src\mbed\tmp\mbed-os-example-wifi" (program)
[mbed] Exec "c:\l\tools\python27\python.exe -u C:\L\src\mbed\tmp\mbed-os-example-wifi\mbed-os\tools\make.py -t GCC_ARM -m UBLOX_EVK_ODIN_W2 --source . --build .\BUILD\UBLOX_EVK_ODIN_W2\GCC_ARM -c -v" in C:\L\src\m
Building project mbed-os-example-wifi (UBLOX_EVK_ODIN_W2, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
[DEBUG] Macros: -DTARGET_STM32F439 -DDEVICE_SPISLAVE=1 -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DTARGET_STM32F439xx -D__FPU_PRESENT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DMBEDTLS_SHA1_C -DTOOLCHAIN_obj
ENABLED -DMBEDTLS_CONFIG_HW_SUPPORT -DTARGET_MODULE_UBLOX_ODIN_W2 -DTOOLCHAIN_GCC -DHSE_VALUE=24000000 -DSUPPORT_80211D_ALWAYS -DTARGET_CORTEX_M -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_U
UPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DMBEDTLS_ARC4_C -DMBEDTL
D_CMSIS_RTOS_CM -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_UBLOX_EVK_ODIN_W2 -DDEVICE_SPI=1 -DCB_INTERFACE_SDIO -DMBEDTLS_MD5_C -DMBED_BUILD_TIMESTAMP=1522236149.65 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDE
Compile [  0.2%]: AnalogIn.cpp
[DEBUG] Compile: C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=
 -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DTARGET_STM32F439 -DDEVICE_SPISLAVE=1 -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DTAR
A1_C -DTOOLCHAIN_object -DTARGET_STM32F4 -D__CMSIS_RTOS -DCB_CHIP_WL18XX -DHSE_STARTUP_TIMEOUT=5000 -DWLAN_ENABLED -DMBEDTLS_CONFIG_HW_SUPPORT -DTARGET_MODULE_UBLOX_ODIN_W2 -DTOOLCHAIN_GCC -DHSE_VALUE=24000000 -DS
M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DTARGET_STM32F439xI -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -DDEVICE_STDIO
DTLS_ARC4_C -DMBEDTLS_MD4_C -DMBEDTLS_DES_C -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -DTARGET_STM32F439ZI -D__MBED_CMSIS_RTOS_CM -DTARGET_FF_ARDUINO -DTOOLCHAIN_GCC_ARM -DTARGET_UBLOX_EVK_ODIN_W2 -DDEVICE_SPI=1 -DCB_INT
DEVICE_PWMOUT=1 -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -DARM_MATH_CM4 @.\BUILD\UBLOX_EVK_ODIN_W2\GCC_ARM\.includes_b5c49d92d9a307d7579d148f7f3b8432.txt -include .\BUILD\UBLOX_EVK_ODIN_W2\GCC_ARM\mbed_config.h -MD
W2\GCC_ARM\mbed-os\drivers\AnalogIn.o .\mbed-os\drivers\AnalogIn.cpp

okoshi3, 2018-03-28:
7.2.1 I tried it with 20170904 (release) but the problem was solved.

Great Thanks.

I did not mean that I was using an old tool for 4.9, but it seems that it is set in MBED_GCC_ARM_PATH.
<code>
C:\L\src\mbed\tmp\mbed-os-example-wifi>mbed compile -c  -m UBLOX_EVK_ODIN_W2 -t GCC_ARM
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
Building project mbed-os-example-wifi (UBLOX_EVK_ODIN_W2, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
</code>
There is no item in the environment variable, but where is it all together?

SeppoTakalo, 2018-03-28:
Can you try `mbed config -L`

```
$ mbed config -L
[mbed] Global config:
CACHE=off

[mbed] Local config (/Users/septak01/src/mbed-os-example-wifi):
TOOLCHAIN=GCC_ARM
TARGET=UBLOX_EVK_ODIN_W2
```

I don't totally understand the settings required. GCC_ARM usually works out of box, so I have not set anything. Also, I'm not using Windows.

okoshi3, 2018-03-28:
Thanks
>mbed config -U -G GCC_ARM_PATH
Complete!

＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿
富士エレクトロニクス㈱ 代表:03-6891-0236
小越博昭               携帯:080-3203-5286
￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣

mikaleppanen, 2018-04-03:
@okoshi3 Is this now solved and can we close this issue?

okoshi3, 2018-04-03:
Yes Closed

＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿＿
富士エレクトロニクス㈱ 代表:03-6891-0236
小越博昭               携帯:080-3203-5286
￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣￣

closed 2018-04-03 by SeppoTakalo