ARMmbed/mbed-os-example-wifi #126: mbed_app_esp8266.json doesn't work with mbed-os-5.10.x

soramame21, 2018-10-12:
**mbed_app_esp8266.json** doesn't work with mbed-os-5.10.x (default) over K64F.

I change mbed_app.json as following. but Tx (D1) doesn't work, which has no output, with ESP8266.
What is the right way to configure esp8266 shield?
```
{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"SSID\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"PASSWORD\""
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "target.network-default-interface-type" : "WIFI",
            "target.components_add"       : ["ESP8266"],
            "esp8266.rx"                            : "D0",
            "esp8266.tx"                            : "D1",
            "esp8266.debug"                         : true,
            "esp8266.provide-default"               : true,
            "nsapi.default-wifi-security"           : "WPA_WPA2",
            "nsapi.default-wifi-ssid"               : "\"xxxxxxxxxx\"",
            "nsapi.default-wifi-password"          : "\"yyyyyyyyyy\""
        }
    }
}
```
`build command:
mbed compile -m K64F -t ARM -f
`

SeppoTakalo, 2018-10-12:
Yes, it seems that https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/mbed_app_esp8266.json is outdated.

At minimum, the line `"esp8266.provide-default"  : true` should be added.

I need to check the pins, but I would have assumed that those have not been changed.

SeppoTakalo, 2018-10-12:
```
WiFi example
Mbed OS version 5.10.1

Scan:
Network: ... <networks removed> ...
15 networks available.

Connecting to mbed...
Success

MAC: 5c:cf:7f:1c:2a:0f
IP: 192.168.100.12
Netmask: 255.255.255.0
Gateway: 192.168.100.1
RSSI: -45

Sending HTTP request to www.arm.com...
sent 58 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

Done
```

Works with following `mbed_app.json`

```
{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"SSID\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"PASSWORD\""
        },
        "wifi-tx": {
            "help": "TX pin for serial connection to external device",
            "value": "D1"
        },
        "wifi-rx": {
            "help": "RX pin for serial connection to external device",
            "value": "D0"
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "esp8266.provide-default" : true
        },
        "UBLOX_EVK_ODIN_W2": {
            "target.device_has": ["EMAC"]
        },
        "NUCLEO_L476RG": {
            "wifi-tx": "D8",
            "wifi-rx": "D2"
        },
        "NUCLEO_F401RE": {
            "wifi-tx": "D8",
            "wifi-rx": "D2"
        }
    }
}
```

I'll modify the app.json and readme.md to fix this issue.

PR #127 (2018-10-12): Fix example configurations for external WiFi shields.
Fixes #126 

@soramame21 For your information.
closed 2018-10-12 by SeppoTakalo