ARMmbed/mbed-os-example-wifi #130: Question about integrating WINC1500 shield driver

kaidokert, 2018-10-27:
@SeppoTakalo  Question, how do we go about integrating a new Wifi shield driver correctly ? We have this shield / module already well working under Mbed 5.9, in progress of upgrading to 5.10. 

However, I'd like to ensure we can make the wrapper compatible with the upcoming TLSSocket class in 5.11 so that we get cloud-client support correctly out of the box, without extra hacking. Note, the module is built very similar to https://github.com/armmbed/wifi-ism43362 , however, it does run full TLS stack on the chip in addition.

The code is not yet 100% ready as some interface functions like get_mac_address() are stubbed out, but TLS connections over https work fine.

SeppoTakalo, 2018-10-29:
There are now couple of things that you need to know:
* `TLSSocket` library is just a easy to use wrapper for Mbed TLS library.
    * It uses Socket API directly. So does not affect any network driver porting.
    * Mbed Cloud Client does not use TLSSocket wrapper. It has its own setup.
* From Mbed OS point of view this  WINC1500 is external IP stack
    * See https://os.mbed.com/docs/v5.10/reference/ip-networking.html
    * Follow porting guide from: https://os.mbed.com/docs/v5.10/porting/networkstack.html
    * And some updates to porting guide from: https://github.com/ARMmbed/mbed-os-5-docs/pull/799
    * Test it by running full set of [Mbed OS Socket tests](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/netsocket/README.md)
* Mbed Cloud Client does not anymore use easy-connect, instead it used default network interface provided by Mbed OS
    * Study the default interface concept from https://os.mbed.com/docs/v5.10/apis/network-interfaces.html
    * Implement configuration option similar to [ESP8266 default option](https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/mbed_lib.json#L16..L18) and [its code](https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/ESP8266Interface.cpp#L579...L584) so you are compatible with Cloud Client.
   * To run Mbed OS socket tests, you also need this implemented, and turn it on by providing `mbed_app.json` similar to [ESP8266 test configuration](https://github.com/ARMmbed/mbed-os/blob/master/tools/test_configs/ESP8266Interface.json)

SeppoTakalo, 2018-10-29:
It short, once the driver is able to pass all Mbed OS socket tests and provides some way of turning on the default WiFi interface, it is compatible with all Socket API users, including Cloud client.

kaidokert, 2018-10-29:
Thanks a lot, that's quite comprehensive. We had missed useful references from here, and we'll work through to bring the code up to speed.
However, this does sound like offloading TLS processing out of the box isnt feasible

SeppoTakalo, 2018-10-30:
The answer for offloading is not so straight forward.

`TLSSocket` is just a wrapper for Mbed TLS. Not a porting interface. Therefore it does not support any TLS offloading to external devices.
In order to offload TLS, you could create your own implementation of `TLSSocket` with a same API and push all the work internally to your module.
This would allow offloading but only if all instances of `TLSSocket` would be replaced with your implementation. So not an ideal, but doable.

Basically if the API is the same, it's just search&replace. So possible, but not without touching the application code.

But because, as I told earlier, Mbed Client does not use `TLSSocket`, this would not help the most important use case for this driver. But it might help some other applications.

SeppoTakalo, 2018-10-31:
Closing the issue as the question have already been answered.

closed 2018-10-31 by SeppoTakalo