ARMmbed/mbed-os-example-wifi #137: NUCLEO_L476RG doesn't work with ESP8266 config

Ronny-Liu, 2018-12-04:
With D8 and D2 connected, wifi connection failed. same configuration works fine with F411.  The print out is:
Connecting to xxxxx...
AT+RST
AT+RST

Connection error: -3012

Ronny-Liu, 2018-12-05:
Target override should also define tx/rx pin in mbed_lib.json of ESP8266 driver.  Suggest to either add this target into ESP8266 driver, or improve 'Getting started' section in readme file, so that user can get it work by following the guide.

michalpasztamobica, 2018-12-17:
@Ronny-Liu , thank you for reporting this issue. Indeed - the declared support for L476RG is not working. Despite trying other UART pins and configs, I could not get this board to work. We decided to remove the support information, as it seems to have been added without a proper verification. We also don't think that this example should support every possible hardware there is. It is meant to be a simple program, using default values as much as possible.