ARMmbed/mbed-os-example-wifi #155: Example does not compile against Nucleo F303RE - IDW01M1 - Wi-Fi module

JanneKiiskila, 2019-04-26:
Using mbed-os-5.12 branch

```
commit eb51a1eeb9799dc1e21d58dde4339a463cc12e7b (HEAD -> mbed-os-5.12, tag: mbed-os-5.12.0-rc4, origin/mbed-os-5.12)
Merge: 089d92b 314e97b
Author: Anna Bridge <Anna.Bridge@arm.com>
Date:   Sun Mar 24 13:45:13 2019 +0000

    Merge pull request #153 from ARMmbed/release-candidate
    
    Updating mbed-os to mbed-os-5.12.0-rc4
```

mbed deploy has been run.

Compiling with 

`jankii01@ubuntu:~/mbed/mbed-os-example-wifi$ mbed compile -m NUCLEO_F303E -t GCC_ARM --app-config mbed_app_idw01m1.json`

or

`jankii01@ubuntu:~/mbed/mbed-os-example-wifi$ mbed compile -m NUCLEO_F411RE -t GCC_ARM --app-config mbed_app_idw01m1.json`

you get this error override.

```
Scan: mbed-os-example-wifi
[ERROR] Attempt to override undefined parameter 'idw0xx1.expansion-board' in 'application[*]'
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/jankii01/mbed/mbed-os-example-wifi"
       Command: "/usr/bin/python -u /home/jankii01/mbed/mbed-os-example-wifi/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F411RE --source . --build ./BUILD/NUCLEO_F411RE/GCC_ARM --app-config mbed_app_idw01m1.json"
       Tip: You could retry the last command with "-v" flag for verbose output
```

JanneKiiskila, 2019-04-26:
It seems you do not include the idw01m1.lib anymore, have you deprecated that configuration (as ST has deprecated the module itself, too). If that's the case, we should delete the app.json -files, too.

JanneKiiskila, 2019-04-26:
@SeppoTakalo

SeppoTakalo, 2019-04-26:
https://github.com/ARMmbed/mbed-os-example-wifi#adding-connectivity-driver

closed 2019-04-28 by JanneKiiskila