ARMmbed/mbed-os-example-wifi #161: [OOB 5.13] Example fails in scan on Nucleo_F411RE + ESP8266

linlingao, 2019-06-07:
The example fails with Nucleo_F411RE+ESP8266. I tried two ESP8266 modules.

WiFi example
Mbed OS version 5.13.0

Scan:
scan() failed with return value: -3012
No WIFI APs found - can't continue further.

mbed compile -m nucleo_f411re -t arm --app-config mbed_app_esp8266.json -c

SeppoTakalo, 2019-06-10:
How did you connect and configure the ESP shield?

Default configuration does not seem to configure pins for F411RE: https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/mbed_app_esp8266.json

-3012 is Device error, so it might be that the device is not in the correct serial port at all.

SeppoTakalo, 2019-06-10:
Oh.. actually, F411RE seems to be using D8/D2 pins by default: https://github.com/ARMmbed/mbed-os/blob/master/components/wifi/esp8266-driver/mbed_lib.json#L59

linlingao, 2019-06-10:
Yes, pins are defined in the mbed_lib.json. No hardware flow control pins are defined though, do you expect it to work without hardware flow control?

SeppoTakalo, 2019-06-10:
It should work without flow control.

What pins did you use?
Please note that TX and RX pins have to be crossed. So the TX pin in ESP is actually Serial RX pin on Mbed side.

linlingao, 2019-06-10:
I'm closing this issue because using the ESP8266 shield has fried my F411RE board. Someone else also mentioned it happened to him too. I don't want to fry another board. I've switched to ST IDW04A1.

closed 2019-06-10 by linlingao