ARMmbed/mbed-os-example-wifi #168: K64F(or NUCLEO_F429ZI) + ESP32

heejungPark, 2019-08-23:
I want to test it I mentioned as title.
So, the esp32-driver(https://github.com/d-kato/esp32-driver) is added to this code.
There are no changes except for `mbed_app.json`.
But I got a not goot result...
What can I do to solve it?

< mbed_app.json >
```
{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"xxxxxx\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"xxxxxxxxxx\""
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines"           : true,
            "platform.stdio-baud-rate"                  : 115200,
            "target.network-default-interface-type"     : "WIFI",
            "esp32.wifi-debug"                          : true
        },
        "K64F": {
            "esp32.wifi-en"                             : "D2",
            "esp32.wifi-io0"                            : "D3",
            "esp32.wifi-tx"                             : "D9",
            "esp32.wifi-rx"                             : "D7",
            "esp32.wifi-rts"                            : "NC",
            "esp32.wifi-cts"                            : "NC",
            "esp32.provide-default"                     : true
        },
        "NUCLEO_F429ZI": {
            "esp32.wifi-en"                             : "D2",
            "esp32.wifi-io0"                            : "D3",
            "esp32.wifi-tx"                             : "D1",
            "esp32.wifi-rx"                             : "D0",
            "esp32.wifi-rts"                            : "NC",
            "esp32.wifi-cts"                            : "NC",
            "esp32.provide-default"                     : true
        }
    }
}
```

< Serial log >
```
WiFi example
Mbed OS version 5.13.0

Scan:
AT? ready%n
AT< ets Jun  8 2016 00:22:57
AT< rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
AT< configsip: 0, SPIWP:0xee
AT< clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
AT< mode:DIO, clock div:2
AT< load:0x3fff0018,len:4
AT< load:0x3fff001c,len:5656
AT< load:0x40078000,len:0
AT< ho 12 tail 0 room 4
AT< load:0x40078000,len:13844
AT< entry 0x40078fc4
AT< .[0;32mI (30) boot: ESP-IDF v3.0.7 2nd stage bootloader.[0m
AT< .[0;32mI (30) boot: compile time 09:04:31.[0m
AT< .[0;32mI (30) boot: Enabling RNG early entropy source....[0m
AT< .[0;32mI (34) boot: SPI Speed      : 40MHz.[0m
AT< .[0;32mI (39) boot: SPI Mode       : DIO.[0m
AT< .[0;32mI (43) boot: SPI Flash Size : 4MB.[0m
AT< .[0;32mI (47) boot: Partition Table:.[0m
AT< .[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length.[0m
AT< .[0;32mI (57) boot:  0 phy_init         RF data          01 01 0000f000 00001000.[0m
AT< .[0;32mI (65) boot:  1 otadata          OTA data         01 00 00010000 00002000.[0m
AT< .[0;32mI (72) boot:  2 nvs              WiFi data        01 02 00012000 0000e000.[0AT< .[0;32mI (87) bootAT< .[0;32mI (95) boot:  5 ota_1            OTA aAT< .[0;32mI (102) boot: End of partition table.[0m
AT< .[0;32mI (107) boot: No factory image, trying OTA 0.[0m
AT< .[0;32mI (112) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x20614 (132628) map.[0m
AT< .[0;32mI (167) esp_image: segment 1: paddr=0x0012063c vaddr=0x3ffc0000 size=0x02d7c ( 11644) load.[0m
AT< .[0;32mI (172) esp_image: segment 2: paddr=0x001233c0 vaddr=0x40080000 size=0x00400 (  1024) load.[0m
AT< .[0;32mI (174) esp_image: segment 3: paddr=0x001237c8 vaddr=0x40080400 size=0x0c848 ( 51272) load.[0m
AT< .[0;32mI (204) esp_image: segment 4: paddr=0x00130018 vaddr=0x400d0018 size=0xdfc80 (916608) map.[0m
AT< .[0;32mI (525) esp_image: segment 5: paddr=0x0020fca0 vaddr=0x4008cc48 size=0x02504 (  9476) load.[0m
AT< .[0;32mI (529) esp_image: segment 6: paddr=0x002121ac vaddr=0x400c0000 size=0x00064 (   100) load.[0m
AT< .[0;32mI (541) boot: Loaded app from partition at offset 0x100000.[0m
AT< .[0;32mI (541) boot: Disabling RNG early entropy source....[0m
AT< 1.1.3
AT< 
AT< I (662) wifi: wifi firmware version: 703e53b
AT< I (663) wifi: config NVS flash: enabled
AT< I (663) wifi: config nano formating: disabled
AT< I (672) wifi: Init dynamic tx buffer num: 32
AT< I (673) wifi: Init data frame dynamic rx buffer num: 32
AT< I (673) wifi: Init management frame dynamic rx buffer num: 32
AT< I (679) wifi: wifi driver task: 3ffdee34, prio:23, stack:3584
AT< I (684) wifi: Init static rx buffer num: 10
AT< I (687) wifi: Init dynamic rx buffer num: 32
AT< I (692) wifi: wifi power manager task: 0x3ffe369c prio: 21 stack: 2560
AT< I (726) wifi: mode : softAP (3c:71:bf:fd:30:5d)
AT< I (733) wifi: mode : sta (3c:71:bf:fd:30:5c) + softAP (3c:71:bf:fd:30:5d)
AT< I (737) wifi: mode : softAP (3c:71:bf:fd:30:5d)
AT(Timeout)
AT> AT+RST
AT? OK%n
AT(Timeout)
AT> AT+RST
AT? OK%n
AT(Timeout)
AT> AT+CWMODE=1
AT? OK%n
AT(Timeout)
AT> AT+CWLAP
scan() failed with return value: 0
No WIFI APs found - can't continue further.

```

heejungPark, 2019-09-04:
Is this case not supported..?
Please inform me any answer.

AnttiKauppila, 2019-09-11:
@heejungPark Why are you opening issue here? Have you asked from d-kato how they have tested and which board? Not directly related to Mbed OS so closing this

closed 2019-09-11 by AnttiKauppila

heejungPark, 2019-09-11:
Sorry, I think that it is related to this example and esp32-driver of d-kato has no problem.
Thanks.