ARMmbed/mbed-os-example-wifi #177: [DISCO-L475VG-IOT01A]  mbed-os-example-wifi search 0 networks available

junyoun-kim, 2019-11-28:
I'm testing this example on DISCO-L475VG-IOT01A board.

And when I flash and test it, it searches 0 networks available even though there are some wifis nearby.

Is there any json settings for this?

michalpasztamobica, 2019-12-04:
Hi @junyoun-kim 

> And when I flash and test it, it searches 0 networks available even though there are some wifis nearby.

What do you mean by "test it". Do you run some mbed-os greentea tests? Do you have your own application? Can you paste the commands you're running to build and run it?

junyoun-kim, 2019-12-04:
I just runed this "mbed-os-example-wifi" example on my DISCO-L475VG-IOT01A board (https://os.mbed.com/platforms/ST-Discovery-L475E-IOT01A/).

And terminal printing below.
------>
WiFi example

Scan:
0 networks available.
No WIFI APNs found - can't continue further.

michalpasztamobica, 2019-12-05:
Hi @junyoun-kim ,

I tried this out on my desk and could not reproduce the issue.

This example will not run on DISCO-L475VG-IOT01A without a slight modification of `mbed_app.json`, as the [README says](https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/README.md#getting-started). This includes adding a `"ism43362.provide-default": "true"`, to [override the driver's](https://github.com/ARMmbed/wifi-ism43362/blob/master/mbed_lib.json#L34) `false` default.
I therefore modified the `mbed_app.json` to this:
```
{
    "config": {
        "wifi-ssid": {
            "help": "WiFi SSID",
            "value": "\"SSID\""
        },
        "wifi-password": {
            "help": "WiFi Password",
            "value": "\"PASSWORD\""
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-convert-newlines": true,
            "target.network-default-interface-type": "WIFI"
        },
        "DISCO_L475VG_IOT01A": {
            "ism43362.provide-default": true
        }
    }
}
```
I added the ism4336 driver library (as suggested in the README for external drivers):
```
mbed add wifi-ism43362
```
Finally I compiled using:
```
mbed compile -t ARM -m DISCO_L475VG_IOT01A
```
And copied the resulting binary onto the chip getting the following output:
```
WiFi example
Mbed OS version 99.99.99

Scan:
Network: systest secured: WPA2 BSSID: B4:5D:50:7d:a7:0 RSSI: -41 Ch: 6
Network: echonet secured: None BSSID: B4:5D:50:7d:a7:1 RSSI: -41 Ch: 6
2 networks available.

Connecting to SSID...
```

@junyoun-kim , is there anything you do differently on your setup that could help me reproduce your issue? Do you have any other board available, that you could use to exclude that this is a hardware issue?

junyoun-kim, 2019-12-05:
Thank you for replying.

Actually I have total 2 devices. But unfortunately both of them fail even with your instructions.. I'm still searching no APs. So do you think I need external antennas for this? Or is there some other steps I can take?

Here is output of mine.(I'm debugging on some logs)
WiFi example
Mbed OS version 5.14.0

        ISM43362: Reset Module
        ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.                                                                                                                                  2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525
Scan:
        ISM43362: End of Scan: cnt=0
scan() failed with return value: 0
No WIFI APs found - can't continue further.

junyoun-kim, 2019-12-05:
oh I got some results. I'll updates soon.

junyoun-kim, 2019-12-05:
I succeed some times. But the probability of success is very low. I succeed one or two times out of 8-9 tries. Do you succeed all times?

michalpasztamobica, 2019-12-05:
Yes, I do.
How far is your WiFi access point? Could it be a matter of range? When your scan works what is the RSSI level that it prints?

junyoun-kim, 2019-12-06:
One WiFi access point is right next to me. And others are also near me.

And I found one weird thing. When I though application succeed to scan, actually ISM43362 debugging logs was printing scan list. But application still found no APs. 

Please refer below logs. Even though ISM43362 succeed to scan and list, at last application saying "scan() failed with return value: 0" I didn't make any modification of codes.

WiFi example
Mbed OS version 5.14.0
_ISM43362: Reset Module
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525_
Scan:
        _ISM43362: received:001,"ureadyintranet",F4:D9:FB:38:2A:6E,-89,144.4,Infrastructure,Unknown,2.4GHz,1
        ISM43362: received:002,"[Cam] ST-2 SJ101U0062",00:00:0F:04:0D:EF,-67,72.2,Infrastructure,WPA2 AES,2.4GHz,1
        ISM43362: received:003,"[Cam] F-CAM-VF-1 1803CR8005645",D4:60:E3:FA:05:60,-62,144.4,Infrastructure,WPA2 Mixed,2.4GHz,1
        ISM43362: received:004,"[Cam] F-CAM-VF-1 1803CR8005658",D4:60:E3:FA:05:6D,-67,144.4,Infrastructure,WPA2 Mixed,2.4GHz,1
        ISM43362: received:005,"[Cam] ST-3 1812CU1000021",74:9D:79:DB:FC:8A,-64,144.4,Infrastructure,WPA2 Mixed,2.4GHz,1
        ISM43362: received:006,"[Cam] ST-3 1903CU1004526",78:B2:13:E2:0B:AC,-59,144.4,Infrastructure,WPA2 Mixed,2.4GHz,1
        ISM43362: received:007,"[Cam] ST-3 1812CU1000067",74:9D:79:DB:FC:B8,-67,144.4,Infrastructure,WPA2 Mixed,2.4GHz,1
        ISM43362: received:008,"dlink-test",00:18:E7:D2:CF:CF,-58,130.0,Infrastructure,WPA2 AES,2.4GHz,1
        ISM43362: received:009,"ureadyintranet",F4:D9:FB:37:77:8E,-74,144.4,Infrastructure,Unknown,2.4GHz,1
        ISM43362: received:010,"iotsystem",3C:37:86:12:53:E2,-47,288.8,Infrastructure,WPA2 AES,2.4GHz,5
        ISM43362: received:011,"iotdapp2g",88:36:6C:D6:22:60,-48,600.0,Infrastructure,WPA2 AES,2.4GHz,4
        ISM43362: received:012,"app3",9C:3D:CF:EA:BF:78,-53,54.0,Infrastructure,WPA2 AES,2.4GHz,5
        ISM43362: received:013,"CellRouter-Sangjin",E4:58:E7:4C:FB:E0,-59,300.0,Infrastructure,WPA2 AES,2.4GHz,5
        ISM43362: received:014,"[Cam] ST-4 EVT000100",B8:DE:5E:4E:D5:80,-62,72.2,Infrastructure,WPA2 Mixed,2.4GHz,6
        ISM43362: received:015,"NETGEAR00",9C:3D:CF:4B:A7:56,-66,216.7,Infrastructure,WPA2 AES,2.4GHz,6
        ISM43362: received:016,"bsp",F8:E9:03:C1:1B:1C,-57,216.7,Infrastructure,WPA2 Mixed,2.4GHz,8
        ISM43362: received:017,"JMhotspot",0A:C5:E1:6E:27:F6,-64,144.4,Infrastructure,WPA2 AES,2.4GHz,6
        ISM43362: received:018,"Carrier 2G",78:D2:94:BE:C6:F2,-68,600.0,Infrastructure,WPA2 AES,2.4GHz,9
        ISM43362: received:019,"setup",F4:D9:FB:37:DD:50,-78,144.4,Infrastructure,Open,2.4GHz,9
        ISM43362: received:020,"ureadyintranet",F4:D9:FB:36:9B:0E,-74,144.4,Infrastructure,Unknown,2.4GHz,9
        ISM43362: received:021,"iotdappapp2g",30:5A:3A:CA:B2:E0,-51,216.7,Infrastructure,WPA2 AES,2.4GHz,9
        ISM43362: received:022,"ureadymobile_2G",F4:D9:FB:37:DD:4D,-78,144.4,Infrastructure,Unknown,2.4GHz,9
        ISM43362: received:023,"power",B0:39:56:E3:98:1A,-43,216.7,Infrastructure,WPA2 AES,2.4GHz,11
        ISM43362: received:024,"IoT_Media",9C:3D:CF:4B:C9:A9,-61,216.7,Infrastructure,WPA2 AES,2.4GHz,11
        ISM43362: End of Scan: cnt=24
        ISM43362: End of Scan: cnt=0_
scan() failed with return value: 0
No WIFI APs found - can't continue further.

michalpasztamobica, 2019-12-06:
@junyoun-kim , I updated my firmware to verify if that could be an issue, but to no avail...

The example does 2 scans. [One is just to return the number of networks](https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L46), and the[ other will actually return the list of APs](https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L57).

I now have the exact same firmware:
```
ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
```
But both scans work for me just fine:
```
ISM43362Interface: read_version = 3525

Scan:

ISM43362: received:002,"mobicaGUESTS",94:9B:2C:18:42:A0,-80,144.4,Infrastructure,WPA2 AES,2.4GHz,1
...[lots of networks here]...
ISM43362: End of Scan: cnt=19
...[lots of networks here]...
ISM43362: received:015,"Mobica",94:9B:2C:18:42:A2,-80,144.4,Infrastructure,Unknown,2.4GHz,1
ISM43362: End of Scan: cnt=15

Network: mobicaGUESTS secured: WPA2 BSSID: 94:9B:2C:2:dd:20 RSSI: -61 Ch: 1
...
[lots of networks here]
...
Network: Mobica secured: Unknown BSSID: 94:9B:2C:18:42:a2 RSSI: -80 Ch: 1
15 networks available.

Connecting to SSID...
```

@LMESTM, @jeromecoutant , can you give us a hand here? Have you seen the behavior that @junyoun-kim  is describing? What more can we check?

@junyoun-kim , can you try to [comment out the first scan](https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L46-L51)  to see if that improves the situation? Or perhaps add some delay between the two scans? Nothing else comes to my mind...

junyoun-kim, 2019-12-11:
Now I just skip wifi scan part(comment out). And success to connect one AP, get IP and other information and disconnect.

Except wifi scan, other works are fine. It seems like I'm in stuck in some point during wifi scanning when scan doesn't work. I don't know exactly. Maybe additional log insert might help to find out scan problem, I don't know exactly where to put in.

michalpasztamobica, 2019-12-11:
It doesn't sound like the problem is specific to this example project. We run this every night in CI on two other platforms and it never failed like this. My local DISCO_L475VG_IOT01A also works fine. 
I don't know if the issue is specific to your board, or the number of WiFi results or anything else. @LMESTM, @jeromecoutant , your help would be much appreciated.

michalpasztamobica, 2019-12-11:
@junyoun-kim , have you tried commenting out **just the first scan**, as I suggested previously? Or perhaps adding a sleep/delay in between the two scans?

junyoun-kim, 2019-12-11:
@michalpasztamobica here is some results.

1. commenting out just the first scan,
Some times second scan succeeds and go to next step. But some times it fails too(return 0 count),

2. adding wait(about 5 seconds) between the two scans.
Some times first scan succeeds and go to second scan. but second scan always fail(return 0 count).
And also first scan fails too some times.

LMESTM, 2019-12-11:
@junyoun-kim 
Would you mind adding further debug logs with debug in ATParser.cpp ?
```
// activate / de-activate debug
#define dbg_on           1
#define AT_DATA_PRINT    1
#define AT_COMMAND_PRINT 1
#define AT_HEXA_DATA     1
```

junyoun-kim, 2019-12-11:
@LMESTM Thank you for reply. Here is failure case log after turn on your debug options.

[failure_log.txt](https://github.com/ARMmbed/mbed-os-example-wifi/files/3949096/failure_log.txt)

LMESTM, 2019-12-11:
@junyoun-kim  Can we also get a log in the case where 1st scan is okay and 2nd scan returns 0 ?

LMESTM, 2019-12-11:
@junyoun-kim actually can you also uncomment the line here:
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ATParser/ATParser.cpp#L331

junyoun-kim, 2019-12-11:
@LMESTM  Okay. Here are fail logs after your instructions.

[fail_at_firstscan.txt](https://github.com/ARMmbed/mbed-os-example-wifi/files/3949164/fail_at_firstscan.txt)
[fail_at_secondscan.txt](https://github.com/ARMmbed/mbed-os-example-wifi/files/3949165/fail_at_secondscan.txt)

LMESTM, 2019-12-11:
@junyoun-kim 
hmm so looks like we're not getting anything back from the module :-( 
If that is confirmed that would be a firmware module limitation.

Let's just confirm that ... could you enable the next debug level here ?
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ATParser/BufferedSpi/BufferedSpi.cpp#L29

If there is a timeout, it should be a 1 minute timeout, do you confirm that it takes about 1 minute to get this 0 scan error ? Or is it somehow immediate ?

LMESTM, 2019-12-11:
Another way of verifying this may be to increase the timeout here: 
https://github.com/ARMmbed/wifi-ism43362/blob/master/ISM43362/ATParser/ATParser.h#L29
You may increase to 60000 for testing purpose and see if that helps

junyoun-kim, 2019-12-12:
@LMESTM I changed timeout to 120000. And it takes about 2 minutes to get 0 scan. So as you guess, it's read timeout when scan have problems. It seems like wifi module problem like you mentioned.

I guess I need to contact the board vendor or the wifi module vendor. Do you have any contact points by any chance to report this issue?

michalpasztamobica, 2019-12-12:
@junyoun-kim , I think it depends on where you bought these boards from... 
You said you have two boards - are they both `DISCO_L475VG_IOT01A` and behaving exactly the same way? Do you have a way to try out another `L475VG_IOT01A` board from a different shop? 
Do you by any chance have `DISCO_F413ZH` available? @LMESTM , please correct me if I'm wrong, but I think it should be using the same WiFi chip.
Perhaps you could check the behavior in a different WiFi environment, for exampe in another office or at home? I wonder if the issue is related to the faulty boards or something strange about the WiFi it is scanning...

korjaa, 2019-12-12:
I think I'm having similar issue with my DISCO_L475VG_IOT01A
```
[14:55:46] Connected
WiFi example
Mbed OS version 5.15.0

Scan:
scan() failed with return value: 0
No WIFI APs found - can't continue further.
```

korjaa, 2019-12-12:
```
WiFi example
Mbed OS version 5.15.0

	ISM43362: Reset Module
AT= ====> ====
AT> I?

AT= ====ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
====
AT= ====OK
====
AT= ====> ====
	ISM43362: get_firmware_version = ISM43362-M3G-L44-SPI,C3.5.2.5.STM,v3.5.2,v1.4.0.rc1,v8.2.1,120000000,Inventek eS-WiFi
ISM43362Interface: read_version = 3525
Scan:
AT> F0

	ISM43362: End of Scan: cnt=0
scan() failed with return value: 0
No WIFI APs found - can't continue further.
```

michalpasztamobica, 2019-12-12:
@korjaa , can you help us by answering the same questions we asked to @junyoun-kim ? Are you also able to see the timeout in communication to the module? Have you tried other boards? Are you able to reproduce this on `DISCO_F413ZH` ? Are you able to try at a different place or different part of the office with other Wi-Fi networks available?

I just tried a few times more, but still - I keep getting the scan report alright every single time. 

@LMESTM , is it possible that different hardware is running on different boards? The firmware version printouts are identical, but perhaps the actual hardware running them is different?

LMESTM, 2019-12-12:
@michalpasztamobica @junyoun-kim 

I checked internally and it turns out this is a known issue : scan in environments with about more than 20 Access Points average (it actually depends on the length of SSIDs) or so would fail :-( 

To workaround this issue, Inventek latest firmware version has a new AT command (F0=2 instead of F0) which list access points 1 by 1. Driver would need to be modified to:
- check if firmware version is the latest one 
- select the AT command accordingly (F0=2 if latest firmware version or newer, F0 if older version)
- list the access points with one of the 2 above methods.

DISCO_F413ZH would have the same issue, or worth as it cannot easily be upgraded to latest firmware version.... 

cc @ABOSTM

michalpasztamobica, 2019-12-12:
That would explain why our CI did not spot the issue - DISCO_L475VG_IOT01A are running in RF-BOXes with only 2 APs available and in my office the scan only brought 19 networks in the results.
@LMESTM , should we move this issue to the [driver repo](https://github.com/ARMmbed/wifi-ism43362)? I have "Transfer issue" button available, but I cannot pick the driver project...

LMESTM, 2019-12-12:
Yes sure, it can be moved there.

michalpasztamobica, 2019-12-12:
I was instructed not to transfer issues, but create a new one instead: https://github.com/ARMmbed/wifi-ism43362/issues/68
Thank you for your help with this, @LMESTM ! Closing the issue, as we cannot really fix it here.

I thought about setting the `count` parameter of the `scan()` method, but looking at the scan implementation, the limit is applied _after_ modem returns the results, which is already too late.

closed 2019-12-12 by michalpasztamobica

LMESTM, 2019-12-12:
> I thought about setting the `count` parameter of the `scan()` method, but looking at the scan implementation, the limit is applied _after_ modem returns the results, which is already too late.

This wouldn't work anyway the firmware already received a request to scan them all and we can't stop it anymore from driver side ... the system is already stuck

junyoun-kim, 2019-12-13:
Thank you for all investigating this issue not giving up. Hope that will fix the issue.

korjaa, 2019-12-13:
I can see ~30 access points with my phone so above explanation makes sense.

korjaa, 2019-12-13:
I wrapped my board in aluminium foil and I'm able to get pass the scan phase now.
[image]