ARMmbed/mbed-os-example-wifi #182: [OOB 5.15] - DISCO_L475VG_IOT01A WiFi connection fails occasionally.

korjaa, 2019-12-13:
With debugs disabled, the example fails for me quite a lot to https://github.com/ARMmbed/mbed-os-example-wifi/blob/master/main.cpp#L98
with error 

```
Connection error: -3011
```

This is a placeholder ticket, I'll add more details as I know.

Having all debugs enabled seems to make the job pass better.

korjaa, 2019-12-13:
I'm seeing like 2 successful connections from 10 attempts.

korjaa, 2019-12-13:
Now during the evening with debug logs enabled, it seems to pass nicely. Not sure if it's the time, or it's debug logs.

korjaa, 2019-12-16:
I made a loop to run the example for the weekend. From the 3880 runs, 3805 succeeded, and 75 failed (1.9 %) with `-3011`. This is with debug logs on.

korjaa, 2019-12-16:
Today I ran the program with no debug logs.
From the 263 rounds, 250 succeeded and 12 failed (4.6%) with `-3011`.

korjaa, 2019-12-16:
The test loop was implemented by sending BREAK in a dummy loop.
``` tcl
#!/usr/bin/expect -f
log_file wifi_loop.log

spawn tio --baudrate 9600 --flow none /dev/ttyACM0 --timestamp
set pid [exp_pid]
send_user "Launched process $pid\n"
sleep 0.5

while true {
	# Reboot board
	send_user "<- sending reboot\n"
	send "\x14"; send "b"
	interact timeout 60 return
}
```
Simple stats can then be extracted with
``` shell
$ grep "WiFi example" wifi_loop.log | wc -l
263
$ grep "Done" wifi_loop.log | wc -l
250
$ grep "\-3011" wifi_loop.log | wc -l
12
```

korjaa, 2019-12-16:
The 250+12=262 != 263 was caused by something weird happening in 11:31:23
```
<- sending reboot
[1;30m[11:31:00] [0mWiFi example
[1;30m[11:31:00] [0mMbed OS version 5.15.0

[1;30m[11:31:01] [0mConnecting to DNA-Mokkula-2G-GJ73fE...
[1;30m[11:31:08] [0mSuccess

[1;30m[11:31:08] [0mMAC: C4:7F:51:05:42:2B
[1;30m[11:31:08] [0mIP: 192.168.1.104
[1;30m[11:31:08] [0mNetmask: 255.255.255.0
[1;30m[11:31:08] [0mGateway: 192.168.1.1
[1;30m[11:31:08] [0mRSSI: -63

[1;30m[11:31:08] [0mDone
[1;30m[11:31:23] [0mWiFi example
[1;30m[11:31:23] [0mMâWiFi example
[1;30m[11:31:23] [0mMbed OS version 5.15.0

[1;30m[11:31:23] [0mConnecting to DNA-Mokkula-2G-GJ73fE...
[1;30m[11:31:26] [0mSuccess

[1;30m[11:31:26] [0mMAC: C4:7F:51:05:42:2B
[1;30m[11:31:26] [0mIP: 192.168.1.104
[1;30m[11:31:26] [0mNetmask: 255.255.255.0
[1;30m[11:31:26] [0mGateway: 192.168.1.1
[1;30m[11:31:27] [0mRSSI: -62

[1;30m[11:31:27] [0mDone
<- sending reboot
```

korjaa, 2019-12-16:
I haven't been able to notice difference between GCC_ARM, IAR and ARMC6.

michalpasztamobica, 2019-12-17:
Hi @korjaa . Have you tried this with any other board (UBLOX, sth+ESP8266, etc.) or another router/network?
Could you also enable `ism-43362`'s logs to see if we can get better information than just -3011 form connect? I would also wonder if the AP has really rejected the connection for any reason, as -3011 stands for `NSAPI_ERROR_AUTH_FAILURE`.... Do you have access to the AP's logs?

korjaa, 2019-12-17:
I'm rerunning the test with NUCLEO_L4R5ZI and ESP8266.

korjaa, 2019-12-17:
The ISM behavior seems to be the same regardless of compiler. Below compilers were tested

- ARMC6.13
- GCC 9
- IAR 8.32.1
- Online compiler

korjaa, 2019-12-17:
Giving the random failures form the ESP, I'm thinking it could be the access point also. I couldn't find any logs form the access point.

michalpasztamobica, 2019-12-17:
If the failure ratio is similar on every device, I would blame the AP, noise, etc. 
We have this [wifi_connect_disconnect_repeat test](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/network/wifi/wifi_connect_disconnect_repeat.cpp). This is running in nightly CI and we didn't get any reports. For nightly runs DISCO_L475VG_IOT01A and other boards are running in an RF-BOX, to reduce the influence of infrastructure on test results.
Could you try if running with the DISCO board available in mervi RAAS with and "RF-BOX" tag will also fail?

korjaa, 2019-12-18:
Switching access point to "mbed" instead of this DNA-Mokkula

NUCLEO_F411RE + ESP8266
From the 59  rounds, 59 succeeded and 0 failed

DISCO_L475VG_IOT01A + ISM
From the 58, 49 succeeded and 9 failed with `-3011`

michalpasztamobica, 2019-12-18:
The last results indicate that this could be an actual SW/FW/chip issue.
Can you please try two more things?
1) Try in an RF-BOX on RAAS (preferably mervi)
2) Enable `ism43362` logs to get better error reporting
Thanks!

korjaa, 2019-12-19:
I was targeting this for today, but unfortunately I ran out of time. I fear this will be pushed to January from my side.

michalpasztamobica, 2020-01-02:
Hi @korjaa , any updates on this issue?

korjaa, 2020-01-03:
Unfortunately not yet, I haven't had enough spare time to get around it.

michalpasztamobica, 2020-01-17:
Hi again @korjaa . Just wondering if you had time to take a look at this issue? I am soon finishing my other tasks and my have some bandwidth to help if you think this issue still needs addressing? As I said, the RF-BOX nightly runs of DISCO_L475VG_IOT01A don't show any sign of frequent failures (at least the few past days that I went through didn't fail even once)

korjaa, 2020-01-20:
Sorry, not yet.

michalpasztamobica, 2020-01-23:
Might be connected to https://github.com/ARMmbed/wifi-ism43362/issues/72

star297, 2020-03-08:
I have not yet been able to connect to any AP's.
Tried various types... Ruckus, Netgear and some Chinese junk.
Tried different AP country settings in case that was an issue. 
I can see some of the AP ssid's but not all.
Stripped down to only one AP and one ssid, with and without (open) security.
Updated and verified firmware as per this:

https://github.com/ARMmbed/wifi-ism43362/blob/master/README.md

Using this 'Official' example:

https://os.mbed.com/teams/ST/code/mbed-os-example-wifi/

Both using the imported OS and updated OS 5.15 (the later version results in a hard fault)

But it simply will not connect.

Before I put this board in the rubbish bin, does anyone have a straight forward 'baremetal' AT driver example that I can try using Mbed - online?

michalpasztamobica, 2020-03-23:
Hi @star297 . Perhaps you could "extract" the [AT commands form the driver](https://github.com/ARMmbed/wifi-ism43362/blob/1978369b2310ea3955715b67869b717fd224a74c/ISM43362/ISM43362.cpp#L436) and use this [SerialThroughput example](https://os.mbed.com/users/sarahmarshy/code/SerialPassthrough/) to talk to the chip directly and see if it scans?
This would remove the example code out of equation.
But yes - it seems your board doesn't work right. Would be good to try out another device.

While I am at this issue report... @korjaa did you by any chance have time to try out the RF-boxed devices, as discusses some time ago?

korjaa, 2020-03-23:
Unfortunately no, I haven't had time to spend on this.

michalpasztamobica, 2020-07-28:
@korjaa , not sure if this issue is still relevant and if it makes sense to keep it open? Please let me know the status.

korjaa, 2020-07-28:
Let's close it.

closed 2020-07-28 by michalpasztamobica